zigpy/zha-device-handlers #4094: [Device Super request] Aqara W100

shennig84, 2025-05-29:
lumi.sensor_ht.agl001

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4447,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0012",
        "0x0402",
        "0x0405",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "Aqara",
  "model": "lumi.sensor_ht.agl001",
  "class": "zigpy.device.Device"
}

quiqueroca, 2025-07-12:
Any update on this feature? @shennig84

williwehrlos, 2025-07-19:
#2

andycastleuk, 2025-07-21:
I've started working on a quirk for this sensor after buying one and realising there wasn't one! 😬  See #4199. 

Still a work in progress as for some reason the sensor isn't reporting battery status and I need to do tests.

kepstin, 2025-08-04:
I have one of these devices and the basics are functional as advertised when using Matter over Thread (temperature reporting with 0.5°C min change; humidity reporting; battery reporting; single/double/hold events for all three buttons), but it looks like the Zigbee mode has some additional features when connected to an Aqara hub:

* Display a "secondary" temperature and humidity value provided by a remote temperature/humidity sensor. (via binding?)
* Display the set point and hvac mode provided by a remote thermostat, which it can also control. (via binding? apparently when a thermostat is bound using the Aqara hub, the buttons control the remote thermostat directly and it no longer allows you to use them to trigger automations)
* Configure the sensor sampling / reporting settings. It looks like you can configure a few different modes, including setting the min temperature difference required for a report, and/or the max time interval between reports.
* Configure the thresholds for the "alarm" and "comfort level" icons on the display.

Hopefully it's possible to figure out how these are implemented so Home Assistant can expose them!

andycastleuk, 2025-08-04:
I need to spend some more time debugging the device, will hopefully get some time this week. When it initially binds it does send additional information, I presume it then sends it intermittently.

The [Zigbee2mqtt library](https://www.zigbee2mqtt.io/devices/TH-S04D.html) has managed to get most things working, not sure about the secondary temperature.

kepstin, 2025-08-05:
Yeah, you can see how the configuration for the buttons, sensor reporting settings, and thresholds for alarm icon displays are set [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/1d81b2d9d951682702ab6ba2148b85b5287299fa/src/devices/lumi.ts#L4571). They also have a function for configuring and sending values for the external temperature/humidity sensor [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/lumi.ts#L2370).

I don't see anything in zigbee2mqtt regarding the battery level reporting or the thermostat setpoint display.

Julian-1-2-3-4-5, 2025-10-19:
how can we get this to be working?

Jebolicious, 2025-10-31:
Maybe as an addition, apparently the W100 has built-in functionality to act as a thermostat as well. Looking at the manual, you have to enable this feature through the Aqara app, which then activates the smaller set temperature reading which can be increased and decreased with the + and - buttons. I hope that we can also read this value in HA, so that I can use these W100's as thermostats in my rooms.

hraschan, 2025-11-23:
@andycastleuk whats your progress so far? Can we help with anything?

Pulzki, 2025-12-06:
> I got your custom quirk working on home assistant yellow, but I cant for the life of me find where the logs end up. I have tried all possible log config and nothing shows upp...
> 
> Is the documentation lacking or how do you get the logs to show?
> 
> I tried:
> 
> ```
> logger:
>   default: info
>   logs:
>     homeassistant.core: debug
>     homeassistant.components.zha: debug
>     bellows.zigbee.application: debug
>     bellows.ezsp: debug
>     zigpy: debug
>     zigpy_deconz.zigbee.application: debug
>     zigpy_deconz.api: debug
>     zigpy_xbee.zigbee.application: debug
>     zigpy_xbee.api: debug
>     zigpy_zigate: debug
>     zigpy_znp: debug
>     zhaquirks: debug
>     custom_zha_quirks: debug
> ```

Should be only to specifiy the name of the file. I got following configuration to work for me:
```yaml
logger:
  default: warning
  logs:
    sensor_ht_agl001: debug
    zhaquirks: debug
```

Pulzki, 2025-12-06:
I've been working on a custom quirk in the mean time of https://github.com/zigpy/zha-device-handlers/pull/4199  that has the functionality to enable/disable the middle display. Still under development but feel free to try it use/modify it.  
Code: https://github.com/Pulzki/w100-zha-quirks

shennig84, 2025-12-14:
Battery Readout is somehow broken
[image]

homerojleal, 2026-02-21:
After the last firmware update 0.0.0_1640 its been having a weird behaviour in HA: 
https://github.com/Koenkk/zigbee2mqtt/issues/27262#issuecomment-3938680947
https://github.com/Pulzki/w100-zha-quirks/issues/9

closed 2026-09-19 by shennig84

homerojleal, 2026-11-10:
> [@homerojleal](https://github.com/homerojleal) Thanks! There is already an open PR ([#5146](https://github.com/zigpy/zha-device-handlers/pull/5146)).
> 
> I haven't opened a PR for my version because preparing and maintaining it properly would take more time than I currently have, and I don't want to open a PR that may not be completed.
> 
> Some of the approaches I use in my custom ZHA quirks differ from those normally used in this repository, so the quirk would need some changes before it could be added here.
> 
> Anyone is more than welcome to use or adapt my quirks for a PR. If I have time, I can also help with the PR or test the changes.

Seems like something broke with HA Core 2026.11 and/or the PR https://github.com/zigpy/zha-device-handlers/pull/5146 - The quirk stopped working.

Has anyone else had this problem?

@nokside - Is the original quirk you wrote still available in any repo/banch/pr?

nokside, 2026-11-12:
> [@nokside](https://github.com/nokside) - Is the original quirk you wrote still available in any repo/banch/pr?

@homerojleal I sold these devices so I can't guarantee that the old version still works.

[aqara_w100.py](https://github.com/user-attachments/files/34759403/aqara_w100.py)

I've reviewed the quirk, and it should still work, although it could be significantly simplified with the current ZHA APIs.

If Aqara eventually fixes the 0.5°C temperature step and supports decimal values in the PMTSD payload, simply replace:

f"T{int(t)}" → f"T{t:g}"

The rest of the temperature handling already supports fractional values.

lukeredpath, 2026-11-12:
I'm still running my fork of @Pulzki's quirk without any issues. I've made a number of improvements and fixes to it with Claude's assistance, particularly around it resetting back to thermostat mode = on (I'm not using it for that, I am using it for its sensors and button automations).

On my branch here:
https://github.com/lukeredpath/w100-zha-quirks/tree/fix/thermostat-mode-reverts

The PR hasn't been looked at so you can download directly from my fork. It's been solid for me since June although I haven't tested it in thermostat mode so YMMV.

I documented the changes I've made here:
https://github.com/lukeredpath/w100-zha-quirks/blob/fix/thermostat-mode-reverts/CHANGES.md

**EDIT**: I decided to pick this up a bit more today and got Claude Fable to review the entire thing for correctness, which it has done. It has written a full test suite and identified various bugs. I've also had it cross-reference with the z2mqtt implementation. If you want to really live on the edge, my updated branch is here. I've also added a Known Issues section to the README.
https://github.com/lukeredpath/w100-zha-quirks/tree/fix/review-findings

**EDIT 2**: As I don't know if these changes will get accepted upstream and I've made so changes from the original, I've gone ahead and merged everything into `main` on my branch. I am happy to accept bug reports on my repo. I've also pushed `1.0.0-beta.1` release.
https://github.com/lukeredpath/w100-zha-quirks/releases/tag/v1.0.0-beta.1

homerojleal, 2026-11-12:
> I'm still running my fork of [@Pulzki](https://github.com/Pulzki)'s quirk without any issues. I've made a number of improvements and fixes to it with Claude's assistance, particularly around it [...]

Cool @lukeredpath  - Do you have this last main branch/beta-release version running currently with your w100 in HA?

lukeredpath, 2026-11-13:
@homerojleal yes, I am running the tagged release but only for a day or so so I can't vouch for its reliability yet. The version I've been running since June was @ `c875fbe53e0d36cde26e624d82e0035b56504c3b` but I was only using it in "thermostat mode = off" so YMMV in thermostat mode.

That said, if you do use either mode I'd highly recommend giving the latest `main` a shot. I'm confident in Fable's ability and it made a lot of fixes with cross-references to the z2m implementation. It also now has very high test coverage. There's a list of known issues in the README and a documented list of the changes I've made in `CHANGES.md`. I did spend quite a bit of time with my device testing the the W100 in thermostat mode and it was working well. The main quirks (no pun intended) I found are:

* Setting the Fan mode to "off" in HA actually sets it to "AUTO" on the device.
* The hardware controls support setting the set point in 0.5 degree increments but the protocol does not seem to support this, so the set point will always get rounded to a whole number. I don't know if this is just a limitation of Zigbee mode and perhaps the device only supports this in Matter over Thread mode?

homerojleal, 2026-11-13:
@lukeredpath - Thanks for sharing and the effort.
I mostly use it as you (sensors + buttons) - But I'm more then willing to use/test a "on the edge" version giving the latest changes in HA ZHA broke some of things in @nokside version (that I also tweeted) and we always have a "basic working stable" version for the basics (sensors + sharing).
If I find any improvements not references in the "known issues" I can/will report them as you suggest. Cheers!

nokside, 2026-11-13:
[aqara_w100.py](https://github.com/user-attachments/files/34830883/aqara_w100.py)

I updated the quirk, and a friend tested it—everything works.

homerojleal, 2026-11-14:
@lukeredpath  - I've test your version and seems stable.
I was using @nokside version also was stable until latest HA-ZHA release.

@lukeredpath - Maybe worth considering a "merged" version between the both quirks, although @nokside is more complete and provides most if not all the entities also considered as "Exposed" written in the https://www.zigbee2mqtt.io/devices/TH-S04D.html specs.

Then maybe we can get one final-stable version into zippy handlers official repo (@TheJulianJES, @mlegner) - This PR https://github.com/zigpy/zha-device-handlers/pull/5146 was a big step back in comparison to the quirks created/improved by @lukeredpath and @nokside, and far advanced already of https://github.com/zigpy/zha-device-handlers/pull/5508.

nstrelow, 2026-11-14:
Hey folks, thanks all for building these. This is super helpful, especially now that it's getting colder outside. Was able to switch my W600 and W100 to Zigbee and use the quirks to get them 99% working how I would like them too. Just the 0.5 temp step problem, but that's a firmware thing from what I understood.
Thanks @nokside @lukeredpath 🙏

lukeredpath, 2026-11-15:
@homerojleal I don't know how practical merging the two implementations is, but it should be fairly trivial to get mine up to parity with the Z2M implementation.

nokside, 2026-11-16:
A quick update: I've created a dedicated repository for my W100 quirk:
https://github.com/nokside/aqara-w100-zha

Although I no longer own the W100, my friends still use it and can help me test updates, so I'll continue maintaining the quirk.

The latest version fixes the E02 error (which didn't affect functionality) and includes some minor code cleanup.

If you encounter any issues or bugs, please report them in the repository rather than in this closed discussion.