zigpy/zha-device-handlers #4099: [BUG] GLEDOPTO GL-SD-001P is being setup as a RGBWW Dimmer

dephcon, 2025-05-31:
### Bug description

Hello,

I've recently added a couple GLEDOPTO GL-SD-001P dimmer relays and they're not detected properly by ZHA. These are just triac dimmers, no RGBWhatever control.

I've manually addressed this in configuration.yaml, but hoping this could be fixing upstream:
```
homeassistant:
  customize:
    light.deck_pergola_lights:
      min_color_temp_kelvin: null
      max_color_temp_kelvin: null
      supported_color_modes:
        - brightness
      color_mode: null
      color_temp_kelvin: null
      color_temp: null
      hs_color: null
      rgb_color: null
      xy_color: null
```

Here's what it is out of the box:
```
min_color_temp_kelvin: 2020
max_color_temp_kelvin: 6329
min_mireds: 158
max_mireds: 495
supported_color_modes:
  - color_temp
color_mode: null
brightness: null
color_temp_kelvin: null
color_temp: null
hs_color: null
rgb_color: null
xy_color: null
off_with_transition: false
off_brightness: 254
friendly_name: Deck Other Dimmer Lights
supported_features: 40
```

Thank you!

### Steps to reproduce

1. add new device or one that's been factory reset via zha

### Expected behavior

1. only brightness controls should be available, besides on/off

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

puddly, 2025-05-31:
Can you upload the device diagnostics?

dephcon, 2025-05-31:
Here you go, apologies, I wasn't aware zha devices even had a diag bundle.

[zha-ce0e30e85baf852070d201e2d4c894c1-GLEDOPTO GL-SD-001P-6cc01a8128fde334ce7fd9c64e479db3.json](https://github.com/user-attachments/files/20533995/zha-ce0e30e85baf852070d201e2d4c894c1-GLEDOPTO.GL-SD-001P-6cc01a8128fde334ce7fd9c64e479db3.json)

dephcon, 2025-05-31:
> puddly left a comment [(zigpy/zha-device-handlers#4099)](https://github.com/zigpy/zha-device-handlers/issues/4099#issuecomment-2925280302)
>
> It looks like the device exposes a Color cluster but sets the capabilities bitmap to 0.

Is that a condition that can be used to identify this specific model and restrict it to brightness only?

I didn't mind the manual override for one unit, but it's so good that I'm going to get a bunch of these.

-------- Original Message --------

dephcon, 2025-05-31:
Is there a way use that as a way to identify it as a non coulor-temp device and not expose the colour temp controls? It's not a big deal, just clutters up the control page

-------- Original Message --------

@simplelogin.co>

GentleHoneyLover, 2025-08-22:
I just bumped into the same issue with a new pair of GL-SD-001P. I've been using two older revisions (GL-SD-001), which are recognized correctly (no color temp) and are using a quirk: `zhaquirks.gledopto.glsd_dimmer.GledoptoGISdDimmer`. The newer ones (GL-SD-001P) aren't using a quirk.

Here's device info for reference:

**GL-SD-001:**
[image]

**GL-SD-001P:**
[image]

Would be great if the newer ones could be recognized correctly out of the box!