zigpy/zha-device-handlers #4102: [Device Support Request] _TZ3000_c7xsiexw 2 gang switch

ausfas, 2025-05-31:
### Problem description

I'm trying to override the default ZHA quirk for a Tuya 2-gang switch (_TZ3000_c7xsiexw, model: TS0002), so I can add support for dp_id 105 (backlight control). Here's the issue:

What I want to achieve

Stop the device from being picked up by the default Switch_2G_GPP quirk
Register both switch endpoints properly (dp_id 1, dp_id 2)
Expose dp_id 105 as a select entity in Home Assistant for backlight on/off

### Solution description

What I’ve tried
Created a v2-style quirk using CustomDeviceV2 and TuyaZHADevice — not picked up

Created a v1-style quirk using CustomDevice and TuyaMCUCluster — picked up and working for switches

Successfully controlled dp_id 105 using zha_toolkit.attr_write

Tried exposing dp_id 105 as a select entity via TuyaDPConverter, but no select entity appears

```
# ts0002_c7xsiexw_manual.py

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Identify, Groups, Scenes, OnOff, Ota, Time
from zhaquirks.tuya import TuyaZBDevice
from zhaquirks.tuya.mcu import TuyaMCUCluster, TuyaOnOffType, TuyaDPConverter

class BacklightDP(TuyaDPConverter):
    """Handle backlight enum (dp_id 105). 0 = off, 1 = normal"""
    values = {
        0: "off",
        1: "normal",
    }
    reverse_values = {v: k for k, v in values.items()}

    def deserialize(self, value):
        return self.values.get(value, f"unknown({value})")

    def serialize(self, value):
        return self.reverse_values.get(value)

class TS0002_C7XS(CustomDevice):
    signature = {
        "models_info": [("_TZ3000_c7xsiexw", "TS0002")],
        "endpoints": {
            1: {
                "profile_id": zha.PROFILE_ID,
                "device_type": zha.DeviceType.ON_OFF_LIGHT,
                "input_clusters": [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE000,
                    0xE001,
                ],
                "output_clusters": [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                "profile_id": zha.PROFILE_ID,
                "device_type": zha.DeviceType.ON_OFF_LIGHT,
                "input_clusters": [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xE001,
                ],
                "output_clusters": [],
            },
        },
    }

    replacement = {
        "endpoints": {
            1: {
                "profile_id": zha.PROFILE_ID,
                "device_type": zha.DeviceType.ON_OFF_LIGHT,
                "input_clusters": [
                    Basic,
                    Identify,
                    Groups,
                    Scenes,
                    TuyaMCUCluster,
                ],
                "output_clusters": [Time, Ota],
            },
            2: {
                "profile_id": zha.PROFILE_ID,
                "device_type": zha.DeviceType.ON_OFF_LIGHT,
                "input_clusters": [
                    Groups,
                    Scenes,
                    TuyaMCUCluster,
                ],
                "output_clusters": [],
            },
        }
    }

    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self._init_tuya_dps()

    def _init_tuya_dps(self):
        tuya_cluster_1 = self.endpoints[1].in_clusters.get(TuyaMCUCluster.cluster_id)
        if tuya_cluster_1:
            tuya_cluster_1.add_dp(1, TuyaOnOffType())
            tuya_cluster_1.add_dp(105, BacklightDP(), force_add=True)

        tuya_cluster_2 = self.endpoints[2].in_clusters.get(TuyaMCUCluster.cluster_id)
        if tuya_cluster_2:
            tuya_cluster_2.add_dp(2, TuyaOnOffType())
```
```
# ts0002_c7xs.py

from zigpy.profiles import zha
from zigpy.quirks.v2 import CustomDeviceV2, EntityType
from zigpy.quirks.v2.devices import TuyaZHADevice
from zigpy.quirks.v2.tuya import TuyaDPMap, TuyaOnOff, TuyaEnum
from zigpy.quirks.v2.homeassistant.switch import SwitchEntity
from zigpy.quirks.v2.homeassistant.select import SelectEntity
import zigpy.types as t

class BacklightMode(t.enum8):
    Off = 0
    Normal = 1

class TS0002_C7XS(TuyaZHADevice, CustomDeviceV2):
    signature = {
        "models_info": [("_TZ3000_c7xsiexw", "TS0002")],
        "endpoints": {
            1: {
                "profile_id": zha.PROFILE_ID,
                "device_type": zha.DeviceType.ON_OFF_LIGHT,
                "input_clusters": [0x0000, 0x0003, 0x0004, 0x0005, 0x0006, 0xE000, 0xE001],
                "output_clusters": [0x000A, 0x0019],
            },
            2: {
                "profile_id": zha.PROFILE_ID,
                "device_type": zha.DeviceType.ON_OFF_LIGHT,
                "input_clusters": [0x0004, 0x0005, 0x0006, 0xE001],
                "output_clusters": [],
            },
        },
    }

    data_point_map = TuyaDPMap(
        TuyaOnOff(dp_id=1, name="switch_1", entity_class=SwitchEntity),
        TuyaOnOff(dp_id=2, name="switch_2", entity_class=SwitchEntity),
        BacklightMode(dp_id=105, name="backlight", entity_class=SelectEntity, entity_category="config"),
    )

    entity_map = {
        "switch_1": EntityType.Member(endpoint_id=1),
        "switch_2": EntityType.Member(endpoint_id=2),
        "backlight": EntityType.Member(endpoint_id=1),
    }
```

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```{
  "manufacturer": "_TZ3000_c7xsiexw",
  "model": "TS0002",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": ["0x0000", "0x0003", "0x0004", "0x0005", "0x0006", "0xe000", "0xe001"],
      "output_clusters": ["0x000a", "0x0019"]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": ["0x0004", "0x0005", "0x0006", "0xe001"],
      "output_clusters": []
    }
  }
}
```

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

ausfas, 2026-05-03:
Finally made it work with backlight
```
from zigpy.profiles import zgp, zha
from zigpy.types import enum8
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    EnchantedDevice,
    TuyaZBE000Cluster,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBOnOffAttributeCluster,
)

from zigpy.quirks.v2 import QuirkBuilder
from zigpy.quirks.v2.homeassistant import EntityPlatform, EntityType

# ---------------------------
# ENUM (this fixes UI + writes)
# ---------------------------
class BacklightMode(enum8):
    Off = 0x00
    On = 0x01

# ---------------------------
# CUSTOM CLUSTER
# ---------------------------
class TuyaZBOnOffAttributeClusterC7XS(TuyaZBOnOffAttributeCluster):
    """Expose 0x5000 backlight attribute as enum8."""

    class AttributeDefs(TuyaZBOnOffAttributeCluster.AttributeDefs):
        backlight_mode = foundation.ZCLAttributeDef(
            id=0x5000,
            type=BacklightMode,
            access="rw",
        )

# ---------------------------
# BASE DEVICE (shared logic)
# ---------------------------
class TS000xBase_TZ3000(EnchantedDevice):
    """Base class containing shared constants for TS000X switches."""

    EP1_BASE = [
        Basic.cluster_id,
        Identify.cluster_id,
        Groups.cluster_id,
        Scenes.cluster_id,
        OnOff.cluster_id,
        0xE000,
        0xE001,
    ]

[... 169 lines cut ...]
BACKLIGHT_MODELS = (
    ("_TZ3000_n1h5w253", "TS0001"),
    ("_TZ3000_c7xsiexw", "TS0002"),
    ("_TZ3000_6bbk8rmq", "TS0004"),
    # Add TS0003 here when known:
    # ("_TZ3000_xxxxxxxx", "TS0003"),
)

for manufacturer, model in BACKLIGHT_MODELS:
    (
        QuirkBuilder(manufacturer, model)
        .replaces(TuyaZBOnOffAttributeClusterC7XS, endpoint_id=1)
        .enum(
            attribute_name=TuyaZBOnOffAttributeClusterC7XS.AttributeDefs.backlight_mode.name,
            enum_class=BacklightMode,
            cluster_id=TuyaZBOnOffAttributeClusterC7XS.cluster_id,
            endpoint_id=1,
            entity_platform=EntityPlatform.SELECT,
            entity_type=EntityType.CONFIG,
            translation_key="backlight_mode",
            fallback_name="Backlight mode",
        )
        .add_to_registry()
    )
```

dfgentag, 2026-09-07:
I have been able to reproduce and identify the relay coupling problem on `_TZ3000_c7xsiexw` / `TS0002`.

## Problem

The device exposes two standard ZCL OnOff endpoints:

- Endpoint 1 → relay 1
- Endpoint 2 → relay 2

However, after joining ZHA, sending an On/Off command to either endpoint causes **both physical relays to switch simultaneously**.

This is not a hardware limitation.

I paired the same device with a Tuya Zigbee gateway and confirmed that both relays operate independently there.

I then captured the complete commissioning process using a CC2531 Zigbee sniffer and Wireshark, comparing the Tuya gateway commissioning sequence with ZHA.

## Zigbee capture / reverse engineering

The Tuya gateway performs an additional commissioning sequence that ZHA does not reproduce exactly.

The relevant sequence observed in the Tuya capture is:

### 1. Basic Read Attributes to APS destination endpoint 255

Cluster:

`0x0000` Basic

Source endpoint:

`1`

Destination endpoint:

`255 (0xFF)`

ZCL Frame Control:

`0x10`

Attributes:

- `0x0004` Manufacturer Name
- `0x0000` ZCL Version
- `0x0001` Application Version
- `0x0005` Model Identifier
- `0x0007` Power Source
- `0xFFFE`

This resembles the existing Tuya "magic spell", but an important difference in the Tuya capture is that this request is sent to APS destination endpoint **255**, rather than endpoint 1.

### 2. Write Basic attribute 0xFFDE

The gateway then writes on endpoint 1:

`0xFFDE = 0x0D`

Datatype:

`uint8`

### 3. Delay

The observed delay between the `0xFFDE` write and the next command was approximately:

`2.55 seconds`

### 4. Basic cluster-specific command 0xF0

The gateway sends to endpoint 1:

`Command: 0xF0`

Observed ZCL payload:

`11 <TSN> F0`

Frame Control:

`0x11`

## Controlled experiments

I reproduced this commissioning sequence in a custom ZHA quirk and tested the individual parts.

Results:

- Normal ZHA commissioning → **both relays coupled**
- `0xFFDE = 0x0D` only → **both relays still coupled**
- `0xFFDE + ~2.55 s delay + exact 0xF0 command` → **both relays still coupled**
- EP255 Basic Read + `0xFFDE` + delay + `0xF0` → **relays became independent**

The significant difference between my unsuccessful test and successful test was adding the Basic Read Attributes request to APS destination endpoint **255**.

I have not yet tested whether the EP255 Basic Read alone is sufficient, so I cannot claim that the other steps are unnecessary.

## Normal operation after commissioning

After successful commissioning, the device can be controlled using normal standard ZCL OnOff commands:

- EP1 controls relay 1
- EP2 controls relay 2

No Tuya MCU datapoints are required for normal relay control.

This was also confirmed from the Tuya gateway capture: the Tuya gateway itself ultimately sends standard On/Off commands to EP1 and EP2.

## Persistence

The independent relay behavior has been successfully tested across:

- multiple Home Assistant restarts
- ZHA reinitialization
- complete removal of mains power from the physical switch

After restoring mains power, the two relays remained independent.

This strongly suggests that the commissioning sequence causes the device to retain the relay mapping/configuration persistently.

## Tested Zigbee device

- Manufacturer: `_TZ3000_c7xsiexw`
- Model: `TS0002`
- Application version tested: `0x52`
- Manufacturer code: `4417`

Observed endpoints:

- Endpoint 1
- Endpoint 2
- Endpoint 242 (Green Power)

The device currently matches:

`zhaquirks.tuya.ts000x:Switch_2G_GPP`

## Commercial product reference

The physical unit used for these tests was purchased from Amazon:

**LOUYSGEN Zigbee Switch Neutral - Blanco**

ASIN:

`B0FD3535Z4`

Product page:

https://www.amazon.com/dp/B0FD3535Z4

I am including the commercial reference only to help identify the tested hardware. As is common with Tuya devices, visually identical products may use different firmware or manufacturer identifiers.

## Working custom quirk

I now have a working custom ZHA quirk that reproduces the Tuya commissioning sequence automatically during pairing.

After pairing with this quirk, both relays become independent and remain independent after Home Assistant restarts and a complete mains power cycle.

I also have the Tuya and ZHA Wireshark/PCAP captures used to compare the commissioning sequences and can provide them if useful.

I would be happy to help test an upstream fix or submit a PR.

ausfas, 2026-11-03:
Thanks @dfgentag for sharing this. I’ve combined your commissioning sequence from #5305 with my backlight implementation from #5007 in the attached custom quirk.

On my _TZ3000_c7xsiexw / TS0002, the Backlight mode dropdown now works. Previously I controlled attribute 0x5000 on endpoint 1, cluster 0x0006, through zha_toolkit.attr_write.

The changes are complementary: yours handles relay commissioning, while mine exposes backlight control. I haven’t independently verified the relay-separation behavior on my installation.

The attached version also applies the sequence to my listed 1- and 4-gang models, but those remain unverified.

I plan to update #5007. How would you prefer to coordinate the two PRs so both features are retained?

[ts0002_c7xs_combined.py](https://github.com/user-attachments/files/34315896/ts0002_c7xs_combined.py)