zigpy/zha-device-handlers #4103: [Device Support Request] Zigbee Water Valve Shutoff (currently works as a "light")

derekcentrico, 2025-05-31:
### Problem description

Currently, the water shutoff valve shows as a light switch.  This can be annoying because it may accidentally get switched off (meaning no incoming water to the house).  

The URL isn't loading for me right now, but I bought it using this link (and several other folks I know with HA + Zigbee are using the same one) https://www.aliexpress.com/item/3256803635243669.html?spm=a2g0o.order_list.order_list_main.10.633d1802XajaZH

I am unsure if there is a way to segregate it from the quirk it's attaching to.

```
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_iedbgyxt",
  "model": "TS0001",
  "class": "zhaquirks.tuya.ts000x.Switch_1G_GPP"
}
```

### Solution description

Separate out the switch off valve to be a different device type, switch or whatever, but not a light.  

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

puddly, 2025-05-31:
Please attach the device diagnostics JSON as well.

derekcentrico, 2025-11-28:
It's a simple merge to add a quirk. The HA log asks for this every reboot.