zigpy/zha-device-handlers #4116: [Device Support Request] Add Startup On/Off Behaviour to Aqara T2 LED Bulb

evakq8r, 2025-06-09:
### Problem description

The Aqara T2 LED bulb supports both a Zigbee and Thread radio, however it appears the Zigbee implementation does not support 'Start-up Behaviour', whereas the Thread implementation on the same light does.

Based on a discussion I had with a user on Discord in the Home Assistant channel (https://discord.com/channels/330944238910963714/1381609970185011260), it appears the functionality exists within an Aqara Hub to change this function, but I do not have a Hub to test with.

Trying to manually change the Zigbee attribute in Home Assistant is not possible as the value is currently reading back as 'None', which I understand to indicate is not implemented.

### Solution description

As it stands, if Home Assistant is not available then the lights will NOT be usable from the physical light switch. The Matter implementation does support this, but I understand Zigbee and Matter/Thread implementations are likely different.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4447,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0102",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "Aqara",
  "model": "lumi.light.agl005",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[zha-0f25044f7abe31f5cf987779ac86d15a-Aqara lumi.light.agl003-43f72e0a09b8800043840c9f7d484d9f.json](https://github.com/user-attachments/files/20656152/zha-0f25044f7abe31f5cf987779ac86d15a-Aqara.lumi.light.agl003-43f72e0a09b8800043840c9f7d484d9f.json)

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

Orange-GT3, 2025-07-02:
Adding support to this request. If useful, here is the signature for the agl003 version:

`{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4447,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0102",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "Aqara",
  "model": "lumi.light.agl003",
  "class": "zigpy.device.Device"
}`

Orange-GT3, 2025-12-30:
Not stale.