zigpy/zha-device-handlers #4125: [Device Support Request] Shelly 1 (PM) Gen 4 Specific Clusters

h1ghrise, 2025-06-13:
### Problem description

With Shelly now supporting Zigbee in their new Shelly 1 (PM) Gen4 devices, there were two specific clusters introduced for handling Manufacturer specific requests.

General Overview of Shelly Devices and their Zigbee capabilities:
- https://shelly-api-docs.shelly.cloud/gen2/Integrations/Zigbee/DeviceFeatures

Clusters:
- Shelly RPC Cluster ([0xFC01](https://shelly-api-docs.shelly.cloud/gen2/Integrations/Zigbee/RPCCluster))
- Shelly WiFi Setup Cluster ([0xFC02](https://shelly-api-docs.shelly.cloud/gen2/Integrations/Zigbee/WiFiSetupCluster)) 

### Solution description

Be able to use those two clusters within the ZHA integration.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

currently, these 2 clusters are not populated:

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 5264,
    "maximum_buffer_size": 108,
    "maximum_incoming_transfer_size": 1613,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 1613,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04"
      ],
      "output_clusters": []
    },
    "239": {
      "profile_id": "0xc001",
      "device_type": "0x2001",
      "input_clusters": [
        "0xfc01",
        "0xfc02"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Shelly",
  "model": "Mini1PM",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

[zha-01JFG7X8TKHW9D9KSYZ6HYAADQ-Shelly Mini1PM-36ba621bb67387c3c88d40509b73f519.json](https://github.com/user-attachments/files/20722531/zha-01JFG7X8TKHW9D9KSYZ6HYAADQ-Shelly.Mini1PM-36ba621bb67387c3c88d40509b73f519.json)

### Logs

no issue - feature request.

### Custom quirk

-

Hedda, 2025-07-18:
> With Shelly now supporting Zigbee in their new Shelly 1 (PM) Gen4 devices, there were two specific clusters introduced for handling Manufacturer specific requests.
>
> General Overview of Shelly [...]

FYI, for reference also check out this related zha discussion here:

- https://github.com/zigpy/zha/discussions/470

PS: That discussion thread maybe be converter into an zha issue?

teodor-hristov, 2025-09-15:
> [@teodor-hristov](https://github.com/teodor-hristov), do you have any input on this? The endpoint's clusters also seem to not be responsive (see [teodor-hristov/Shelly-Zigbee-Tools#1](https://github.com/teodor-hristov/Shelly-Zigbee-Tools/issues/1)).
> 
> Is there a way to bring this up with Shelly directly?

Hey, sorry for missing this issue!

All current Shelly devices expose their custom clusters on endpoint 239, as described in our API docs. If I remember correctly, you already added a quirk or some overriding code for handling these clusters.

From what I can see, each operation reaches the device and the device processes it, but the response either doesn’t come back or something unexpected happens. Is this the same on your side? Are you able to use those clusters successfully? Are you able to see the ShellyWifiCluster and ShellyRPCCluster?

Hedda, 2025-09-18:
> I would be happy to contribute to ZHA / zha-quirks once I get my head wrapped around a possible solution and got some proof of concept running.

@kaktus42 that sounds good, please feel free to join the discussion in zigpy's zha repository if and when you do, even if it just to add some feedback and input or ideas:

- https://github.com/zigpy/zha/discussions/470

Hedda, 2026-01-16:
@teodor-hristov you previously wrote above that you have some internal zha_quirks for those clusters: 

_Originally posted by @teodor-hristov in https://github.com/zigpy/zha/discussions/470#discussioncomment-14623696_

> _**We have our internal zha_quirks for those clusters; they are pretty unprofessional because we only wanted to test. Are they going to help you?**_

Does those include Wi-Fi Configuration? 

Could you maybe regardlless share those internal zha_quirks for Shelly WiFi?

PS: Again, there is also more discussion about that and more here -> https://github.com/zigpy/zha/discussions/470

teodor-hristov, 2026-01-16:
> [@teodor-hristov](https://github.com/teodor-hristov) you previously wrote above that you have some internal zha_quirks for those clusters:
> 
> _Originally posted by [@teodor-hristov](https://github.com/teodor-hristov) in [zigpy/zha#470 (reply in thread)](https://github.com/zigpy/zha/discussions/470#discussioncomment-14623696)_
> 
> > _**We have our internal zha_quirks for those clusters; they are pretty unprofessional because we only wanted to test. Are they going to help you?**_
> 
> Does those include Wi-Fi Configuration?
> 
> Could you maybe regardlless share those internal zha_quirks for Shelly WiFi?
> 
> PS: Again, there is also more discussion about that and more here -> [zigpy/zha#470](https://github.com/zigpy/zha/discussions/470)

Yes, let me ask our colleagues about that.

GabboPenna, 2026-11-07:
@teodor-hristov, a related firmware issue may be useful to the Shelly team: Gen4 relay firmware 2.0.x and 2.1.1 can keep sending reports while all directed Zigbee traffic fails at APS delivery. It is reproduced across Zigbee2MQTT and ZHA, with TI and Ember coordinators: Koenkk/zigbee2mqtt#32704.

I confirmed it on six `Mini1PM` devices running 2.1.1. ZHA keeps them available because telemetry continues, but reads and On/Off commands can return `ZIGBEE_DELIVERY_FAILED`; Wi-Fi remains reachable and LQI is good.

I opened zigpy/zha#1004 for the controller-side health blind spot and #5669 to warn ZHA users about the affected firmware. Could you confirm whether Shelly has a firmware fix or test build planned? I can validate it on one device and provide sanitized ZHA logs.

teodor-hristov, 2026-11-07:
> @teodor-hristov, a related firmware issue may be useful to the Shelly team: Gen4 relay firmware 2.0.x and 2.1.1 can keep sending reports while all directed Zigbee traffic fails at APS delivery. It [...]

We were trying to reproduce this issue but we can't. Can you please explain how to reproduce it? Can you plese explain in depth what configuration do you use for the shelly, is it still reproducible when wifi is disabled?

GabboPenna, 2026-11-07:
Thanks for looking into it. The important detail is that this has not been an immediate or deterministic post-join failure in my setup. The devices initially work and then degrade in place after normal operation, sometimes after several days. A reboot or rejoin can therefore make a short test look successful.

My setup is:

- Six Shelly 1PM Mini Gen4, model `S4SW-001P8EU`, app `Mini1PMG4ZB`
- Stable firmware `2.1.1` (`20261026-151139/2.1.1-g07e9ad2`)
- Both radios enabled: Wi-Fi STA connected and `zigbee.enable: true`
- Shelly Cloud enabled, MQTT disabled
- Input type `switch`, output mode `follow`, initial state `match_input`; timers disabled
- Home Assistant Core 2026.11.0, ZHA 2.4.2, zigpy 2.4.1, bellows 1.2.1
- SONOFF ZBDongle-E connected directly over USB, EmberZNet 6.10.3.0, Zigbee channel 25
- 34-node network: 24 routers and 9 end devices

Observed reproduction sequence:

1. Join the device to ZHA and configure Wi-Fi as well.
2. Use it normally for an extended period. Initially Zigbee reads and On/Off commands succeed.
3. Eventually the device continues to send unsolicited electrical measurement/metering reports, retains its NWK address, reports `network_state: joined` over RPC, and remains reachable over Wi-Fi.
4. Send a directed On/Off command or Basic/OnOff cluster read from ZHA.
5. The request may then fail after the APS retries with `sl_Status.ZIGBEE_DELIVERY_FAILED: 3074`. During the same interval, incoming reports continue and ZHA keeps the device available with fresh `last_seen` data.
6. Rebooting or rejoining restores bidirectional Zigbee temporarily, after which the state can recur.

During my capture, five directed requests failed across two devices while their incoming reports continued. Their LQI values were 108-168, and the coordinator logged no reset, serial, queue, or buffer errors. One currently inspected unit has more than 11 days of uptime, Wi-Fi RSSI -63 dBm, and still reports Zigbee as joined.

I have not yet run a controlled long-term test with Wi-Fi disabled, so I do not want to claim a result for that case. I can dedicate one unit to an A/B test. Could you clarify the condition you want tested: disabling STA with `WiFi.SetConfig` while retaining its configuration, or factory-resetting/configuring the unit as Zigbee-only? Please also let me know which debug components or logs would be most useful from the Shelly side. I can run the test for several days and compare it with an otherwise identical unit with both radios enabled.

teodor-hristov, 2026-11-07:
> Thanks for looking into it. The important detail is that this has not been an immediate or deterministic post-join failure in my setup. The devices initially work and then degrade in place after [...]

Just set wifi to disabled using wifi.setconfig and reboot. That's it. Coexistence is the most performance dependant thing in these devices.

GabboPenna, 2026-11-07:
Understood, thank you. I will disable STA with `WiFi.SetConfig`, reboot one Mini1PM, and leave an otherwise identical 2.1.1 unit with Wi-Fi enabled as the control. I will avoid any further reboot or rejoin during the observation window and report the directed-request and unsolicited-report results here after several days.

paveljandejsek, 2026-11-07:
I _think_ I am sometimes having the same or at least very similar issue with 1 mini (running zigbee only with wifi disabled), it seems to be totally random on which unit and when it will occur (I had it like three times on one unit, but I also have couple ones that had no issue at all as of yet), so it is not just related to one unit and I always fix it by restarting it (in the past both enabling wifi and restarting from UI worked as well as just sending Shelly.Reboot RPC through zigbee)

Anything I should do if/when it occurs again?

GabboPenna, 2026-11-07:
The A/B test is now running (started 2026-11-07 19:42 CET).

Test setup:
- Shelly Mini 1PM Gen4, firmware 2.1.1
- ZHA on Zigbee channel 25
- Wi-Fi STA disabled with `WiFi.SetConfig`, followed by a reboot, on one test device
- the other five dual-radio Shelly Gen4 devices remain unchanged as controls
- a local recovery script will restore Wi-Fi and reboot the test device after 14 days

Initial result: the Wi-Fi RPC endpoint became unreachable as expected, the relay output remained on, and ZHA received the first post-reboot voltage report 38 seconds after the test started. Further voltage/frequency reports followed, with no related ZHA/zigpy errors in the checked log window.

I will keep it in Zigbee-only mode and report back with reporting continuity and command reliability compared with the control devices.

GabboPenna, 2026-11-11:
Checkpoint for the `WiFi.SetConfig` test requested above:

- All six Mini 1PM Gen4 units are on stable firmware 2.1.1 with Wi-Fi STA disabled.
- They have now run for 83.75 hours since the full-group rollout without a reboot or Zigbee rejoin.
- Home Assistant recorded 359,853 voltage samples, 0 unavailable/unknown samples, and a maximum report gap of 72.3 seconds across the group.
- A new uncached On/Off attribute read succeeded on all six devices; response times were 0.074-2.992 seconds.
- The retained ZHA/zigpy log windows contain no timeout, retry, delivery-failure, or unavailable match for the group.

I have therefore not reproduced the one-way failure in the Wi-Fi-off group so far. I do not consider this proof that coexistence is the root cause: I moved all six units to Wi-Fi-off after the initial check, so there is no longer a simultaneous control, and other users have reported the same failure with Wi-Fi disabled. I will continue the planned 14-day observation.

The Zigbee2MQTT thread now says Shelly is testing a fix. If a candidate firmware is available, I can put it on one unit and compare it with the remaining five using the same checks.