zigpy/zha-device-handlers #4128: [BUG] Philips Hue Motion Sensor (SML003) battery always at 100%

Sawtaytoes, 2025-06-16:
### Bug description

The battery reporting for the newer Philips Hue Motion Sensors (SML003) is always showing 100% via ZHA. The device never shows anything lower. I've had devices go offline because their battery depleted with no change to the reported battery remaining.

### Steps to reproduce

1. Pair a newer Philips Hue Motion Sensor (SML003) with ZHA.
2. Open the ZHA device in Home Assistant.
3. Leave it on for a few months.
4. No change in the battery levels. Still shows 100%.

### Expected behavior

I expect that the battery will track with the actual battery available in the two AAA batteries. I wanna know when the device will suddenly stop working.

There's also a point where the device doesn't work with Low sensitivity, but Medium and higher still work. I'd like to know at what point that occurs.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

MattWestb, 2025-06-18:
Battery voltage is normally not being used only for bad Aqara devices and some more that is not reporting % and ZHA is calculating % from it.

The reconfigure looks OK and the battery % renaming shall being reported OK if device is doing it right.

Can you reading the attribute `battery_percentage_renaming` by from the device-card the tree dots (`...` ) right of  r`econfigure` and then `manage Zigbee device`,  `cluster`, `PowerConfiguration`, `Attributes` and select the `battery_percentage_remaining` and click `read attribute` and waking up the device so it can getting the command.
PS: The reading is X2 so if getting 120 its the real value 60%.

Sawtaytoes, 2025-06-18:
[image]

There are 2 batteries, showing numbers 2 and 3.

I tried reading some other values, but this is the only other non-"None":

[image]

TheJulianJES, 2025-06-24:
I have an SML003 that reports 79 % (`battery_percentage_remaining=157`) and 2.7 V (`battery_voltage=27`).
The reporting is straight from the device. I've also never had any Hue sensors die on me. Most of them are already in place for multiple years and they're all still going strong.