zigpy/zha-device-handlers #4137: [Device Support Request] Aqara Vibration Sensor lumi.vibration.agl01

jorgenbosman, 2025-06-19:
### Problem description

I just got myself a new Aquara vibration sensor lumi.vibration.agl01 (DJT12LM). It is detected by ZHA, but Configuration takes a very long time and no useful sensors are added, only "IAS Zone" whatever that may be... So I assume that there is no support for this model yet in ZHA? Can this be added please?

### Solution description

Support for the new zigbee 3.0 aqara vibration sensor

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

cnotin, 2025-12-17:
I have started to work on this and found some interesting leads. I've never written a ZHA handler though so I'm not progressing very fast. If anyone more experienced want to team up please let me know!

cnotin, 2026-08-25:
> I don't think the 2.0 migration changed anything regarding the sensitivity. I wonder if I get the aqara hub and set the sensitivity to high, would that be preserved if I reset the sensor and pair it back to HA? I have no intentions to use their hub constantly, just for the sensitivity settings.

@Hakun1n if you have access to it it'd be interesting to see if it works! But also see what @mengwong shared on that test in https://github.com/zigpy/zha-device-handlers/issues/4137#issuecomment-4205558840

> also thanks to the tape (it is not hard-fixed but it can wobble a bit)

It's fragile and weird but I guess it works due to the wobbling indeed 😅

> Maybe 3-5sec? I assume i can just change it in the code and restart HA (not sure if this is something which could be configurable from HA device page?)

I agree and I thought about it... I'll try!

> Anyway, thanks for your effort and time spent on this. Even though it’s not perfect, you’ve made at least one person happy, hope there will be more once it gets merged with the main :)

I'm very happy to hear that as well! Thanks for sharing your experiments and results too :)

cnotin, 2026-08-25:
Update: last commit of [PR #4972](https://github.com/zigpy/zha-device-handlers/pull/4972) adds a ZHA UI setting for the coordinator-side vibration reset timeout (default 70 seconds, persisted across Home Assistant restarts). For an existing installation, replace `/config/zha_quirks/vibration_agl01.py` and reload ZHA; no device re-pairing is needed. This addresses timeout behavior, while the vibration-sensitivity question above remains separate and is not exposed by this PR yet.

cnotin, 2026-08-25:
PR #4972 now includes an experimental sensitivity select for the DJT12LM, using the T1-specific Aqara attribute rather than the older DJT11LM command.

Could someone with this sensor please test whether sensitivity really changes? Readback does not work for this attribute, so we cannot confirm the setting by reading it from the device: set an explicit value, then compare the same physical vibration at High and Low.

Because the sensor sleeps, briefly press its physical button immediately before changing the sensitivity. It may also be worth trying a button press immediately after the change. I am not yet certain that the physical sensitivity changes on my device, so if anyone finds a repeatable procedure or observes a clear difference, please share the result.

Hakun1n, 2026-09-07:
@cnotin Missed the last update for whatever reason, but I just got some PR update notifications and checked here as well. I'll try to do some tests and report back in a day or two.

Hakun1n, 2026-10-02:
@cnotin Hey, sorry for the delayed feedback. I did some tests with the latest version of your quirk. I have two of the T1 sensors. One under the kitchen top and second I am not using so I could play with it a little on my desk.

I had the kitchen one set to Low/High and Low sensitivity again and used it for a week for each. I haven't noticed any significant difference. I am only using it for the tripple-tap to turn on/off the lights and here I don't think the sensitivity matters. See below.

For the other one I did some tests. Since I don't have any measuring device I used plastic top/lid from small Ikea box and placed the sensor on top of it. This makes the sensor elevated a bit and since the plastic is quite soft it is easy to "tap" or move/wiggle a bit to trigger the detection. I was testing the Low/High settings. Skipped the Medium since the difference between L/H is quite small so the Medium in between is hardly distinguishable w/o real measuring tool.

The (subjective) conclusion is... 
- **Tripple-tap** - I don't see any difference. The same place/intensity gets registered on Low and High. It works in both modes until it does not in either of them. The good thing is that very slight tripple-taps are still registered.
- **Vibration** - Here the tapping on the lid did not worked consistently. Seems like the sensor needs more "constant" vibration. So i was moving the whole lid back and forth a little. And here I could actually notice difference between Low and High settings. On Low i had to move the lid approx by 1cm back and forth to get the vibration registered. On High it takes only like 3mm or so to trigger it. Medium is hard to say, something between.

Meaning, the **sensitivity setting works**, but the sensor is quite sensitive on its own so the difference is very small. 
Then only place where I could see some difference was when I placed the sensor on my printer and started printing random stuff while the reset timer was set to 1. On high it was constantly detected. On Low the detection dropped more often.
I also tested the sensor on my wash-machine. It has quite good dampers but the sensor was detecting the Draining cycle even on Low sensitivity.

The real game-changer for me is the reset timeout setting so i can fine-tune the automation w/o restarting the whole HA every time I change the value.

[image]

Hope this helps !

cnotin, 2026-10-02:
Thanks for sharing such a detailed feedback! And I love your experiments 👌
Coincidence is that I moved mine yesterday from the washing machine, where I didn't manage to achieve an acceptable result (but your experience makes me hesitate to retry!), to a countertop and I played with the sensitivity on triple-tap and indeed I didn't notice any change.
Anyway, thanks for confirming that the sensitivity setting works at least for single taps. I'll keep it in the quirk.

Hakun1n, 2026-10-02:
Well, it "likes" side-to-side movements rather than "taps" on top (or bumps up and down). Single-tap gets hardly registered. Also seems like the detection is not the best when it is in vertical position. I'll try different places and orientation next time the wash-machine is doing its magic, maybe I was just lucky to place it on ideal spot :)

cnotin, 2026-10-02:
What a capricious pet :)

Hakun1n, 2026-10-02:
I just noticed the weird battery reporting. Both of the sensors doing it so maybe it is just a thing of the device itself?

Maintenance dashboard shows yellow (i think i also noticed 10%-red some days ago but thought it was some glitch)
[image]
[image]
</br>

In reality it just fluctuates up and down, even though nothing is waking up the kitchen sensor for at least half of the day/night and the other sensor was sitting on the shelf for several days
[image]</br>

[image]

cnotin, 2026-10-02:
I observe it too on mine. But I have other sensors doing the same so I wasn't surprised even though I agree that it isn't normal.
No idea yet for the reason sorry...