zigpy/zha-device-handlers #4144: Add quirk for Tuya TS0601 soil sensor (_TZE200_npj9bug3): Fix soil moisture mislabeling & expose missing air humidity

Arashrk82, 2025-06-23:
Hi! I'd like to request proper ZHA support for a Tuya TS0601-based soil monitoring sensor.

Model details:

Manufacturer: _TZE200_npj9bug3

Model: TS0601

IEEE: a4:c1:38:b5:11:c0:d8:cb

Power source: Battery

Quirk applied: No

Expected sensor entities:

🌡️ Air temperature (device_class: temperature)

💨 Air humidity (device_class: humidity)

🌱 Soil moisture (device_class: moisture)

🔋 Battery level (device_class: battery)

Current behavior in ZHA:

One temperature sensor — seems to work correctly (likely air temp)

One humidity sensor — this is actually soil moisture

No air humidity sensor is exposed

Battery sensor works

No quirk is currently applied

Tested behavior:
I confirmed the "humidity" sensor is soil moisture by inserting the probe into water — values jumped to 80%, confirming it's soil-based, not ambient air.

What’s needed:

A ZHA quirk that:

Properly decodes all 3 sensor readings from Tuya’s custom cluster (air temp, air humidity, soil moisture)

Maps soil moisture to device_class: moisture

Maps air humidity to device_class: humidity

Ensures these entities are correctly named and exposed

Diagnostics:
I have the full HA diagnostic .json for this device attached.
Happy to help test a proposed quirk.

Thanks a lot!

[zha-01JKEB6Y059TR72MGYSTRKXH8M-_TZE200_npj9bug3 TS0601-005aa5000a31519d02675027ee0098b1.json](https://github.com/user-attachments/files/20872588/zha-01JKEB6Y059TR72MGYSTRKXH8M-_TZE200_npj9bug3.TS0601-005aa5000a31519d02675027ee0098b1.json)

Arashrk82, 2025-06-23:
**Additional info for comparison:**

I also have another TS0601 sensor:

- Manufacturer: `_TZE284_aao3yzhs`
- Model: `TS0601`

This one only has temperature + soil moisture + battery (no air humidity sensor), and **works correctly with ZHA**. It shows the moisture reading with `device_class: moisture` and works in the Plant integration.

The problematic sensor (`_TZE200_npj9bug3`) seems to include air humidity, but it’s not exposed, and its soil moisture is mislabeled as "humidity".

Let me know if you want diagnostics from the working sensor too.

Arashrk82, 2025-06-24:
Update (confirmed via Zigbee2MQTT):

I temporarily tested my _TZE200_npj9bug3 soil sensor in Zigbee2MQTT. It reports all values correctly, with the following fields:

json
Copy
Edit
{
  "temperature": 29.7,
  "humidity": 58,
  "soil_moisture": 78,
  "battery": 100
}
So the sensor exposes:

temperature → air temperature

humidity → air humidity

soil_moisture → actual soil moisture (not air humidity!)

battery → battery %

This confirms that ZHA is currently misclassifying soil_moisture as humidity and not exposing air humidity at all.

Zigbee2MQTT seems to be using DP mappings:

3: Soil moisture

5: Temperature

109: Humidity

Hope this helps with creating a quirk for proper support.

[image]
[image]

crstian19, 2025-09-06:
> [@Arashrk82](https://github.com/Arashrk82) would you be willing to try a quirk?

I can confirm this quirk works for me! Thanks a lot @makuser !

[image]

intra-aud, 2025-11-26:
> Will this work for a similar device - "TS0601 by _TZE284_o9ofysmo" - a soil sensor?

Howdy,

I also have the same soil sensor, looking for some help to write a quirk or if someone can direct me to one that already works, that would be excellent!