zigpy/zha-device-handlers #4149: [Device Support Request] Добавить поддержку мотора для жалюзи TS0601 _TZE284_2gi1hy8s

Kirill172, 2025-06-25:
### Problem description

Устройство
TS0601
_TZE284_2gi1hy8s
Подключено через [Generic Zigbee Coordinator (EZSP)](http://192.168.100.82:8123/config/devices/device/c582ae3cd3d75ddd51e30b4fe88bafb5)
Zigbee info
IEEE: a4:c1:38:fa:b8:52:f7:24
Nwk: 0x9fef
Device Type: EndDevice
LQI: 236
RSSI: -41
Устройство было в сети: 2025-06-25T22:52:24
Источник питания: Battery or Unknown
https://www.ozon.ru/product/zigbee-umnyy-motor-dlya-rolstavney-podhodit-dlya-rulonnyh-shtor-zhalyuzi-golosovoe-upravlenie-alice-2080621114/

### Solution description

I would like support for the TS0601 device by _TZE200_gkfbdvyx in ZHA.
Currently, after pairing, the device does not expose any sensors or entities in Home Assistant.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Diagnostics information

[zha-ed6236a22b5a604d8f7f800c42ed3c34-_TZE284_2gi1hy8s TS0601-5c3b8501b67700d6776a88a2601999a7.json](https://github.com/user-attachments/files/20910470/zha-ed6236a22b5a604d8f7f800c42ed3c34-_TZE284_2gi1hy8s.TS0601-5c3b8501b67700d6776a88a2601999a7.json)

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

Bespinuk, 2025-07-03:
i would also like this added. thanks

SquareFaceBE, 2026-10-10:
Got the same motor working here and dug through some debug logs. Two things I haven't seen mentioned yet.

**Battery sits on DP 13.** Value type, 0 to 100. The number might have to be doubled. The motor only sends it right after a movement finishes, so if you're watching an idle device it looks like there's nothing there at all. Mine drops about a percent per full open-close cycle.

Charging is a different story. Plug it in and DP 13 comes in every 56 seconds, one percent up each time, and it stops the second you pull the cable. I couldn't find a dedicated charging datapoint, but that pattern is regular enough to build a charging sensor on if you want one. 

**Position might needs the inverted flag on this model.** I suspect this is what people here are hitting when they say position doesn't work. The motor reports 0 when the curtain is fully open.

DP map as I saw it:

| DP | Type | Function |
|---|---|---|
| 1 | enum | 0 = open, 1 = stop, 2 = close |
| 8 | value | current position, 0 = fully open |
| 9 | value | target position 0-100 |
| 13 | value | battery percentage 0-100 |

Hope someone is helped with this.