zigpy/zha-device-handlers #4159: [Device Support Request] TZ3210_zana8d08 - Smart Curtain Switch Module MOES

luismpmatos, 2025-07-01:
### Problem description

Hi 👋 

I recently installed a MOES smart curtain mini module (https://moeshouse.com/products/zigbee-smart-curtain-switch-module-mini-design-app-remote-control?variant=50187664097595) recognized as a Tuya TS130F but I am facing some issues with calibration, changing the switch type from reset to rocker switch and also the percentage indicator bar that seems to be reversed. 

I have the bigger sibling of this module, the Moes | MS-108ZR, which I was able to calibrate using the Tuya endpoint via calibration parameter, however in this case there is no such endpoint but rather a WindowCovering endpoint with no calibration parameter.

Would appreciate if someone could help out with this integration, thanks in advance!

### Solution description

Possibility to change the module switch type, being able to conduct some calibration of the opening/closing functions and also displaying correctly the open/close bar percentage. It

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102",
        "0xe001",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_zana8d08",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

JPT77, 2025-07-06:
same here. No idea how to calibrate the Moes. 
Homeassist shows TS130F, TZ3000. Case reads MS-108ZR
Why does it identify as the TS130F if it obviously is a completely different device?

mine looks totally different from the above link
https://moeshouse.com/collections/curtain-switch-module/eu

now my shutter sit's at 1/3 and can go down another 1/3 and back ub 1/3. It just doesn't know the full length of the shutter.
This occurred after rebooting the server running Homeassist, not sure if the reboot is related.

is there anything I can help with?

luismpmatos, 2025-07-06:
Hi @JPT77 I also have two like yours currently working fine. I followed the instructions in this post (https://community.home-assistant.io/t/zha-curtains-module-calibration/359996/5?u=lmpmatos) to calibrate the modules properly, give it a try ;).