zigpy/zha-device-handlers #4177: Is the Tuya Motion ZG-204ZM supported in ZHA or not?

schabau, 2025-07-09:
Hello,

According to what I've found here, the Tuya Motion ZG-204ZM is supported in ZHA. That's why I bought it. That's why I bought it.
However, it doesn't seem to work properly in ZHA. There's no configuration, and the sensors available are lighting, and apparently only PIR for motion. In Z2M, you can switch motion detection between PIR, PIR+Radar, and Radar, among other options. Nothing works in ZHA.

Do I have a different model?

Best regards,
Detlef Paschke

[image]

[image]

rinyheuvelman, 2025-07-12:
I bought some of those sensors on AliExpress at different stores. One integrated as expected and showed up as a pir and mmWave sensor with the options of adjusting the sensitivity and distance. Couple of others turned up as hobeian and doesn’t show the mmWave options although the sensor has a mmWave sensor inside (I’ve opened up two of them). 
It seems like the configuration in ZHA is not working correctly. Also the code on the board is ZG-204ZS instead of 204ZM

[image]

schabau, 2025-07-15:
Hello,
I've already done the same thing with my Girier ME168 thermostats ("_TZE200_cxakecfo") as I'm doing here with the HOBEIAN ZG-204ZM. I downloaded the tuya_motion.py file here and added the corresponding line (in this case, line 1439).

`.applies_to("HOBEIAN", "ZG-204ZM")`

The sensor now works with ZHA, although I believe that radar is only faked in this model. The ZG-204ZS model only shows presence in both ZHA and Z2M when you move. If you sit still in front of the sensor, there's no detection.

Best regards,
Detlef Paschke

rinyheuvelman, 2025-07-15:
Just for the record: I assume you added the python file to the zha quirks folder and in your configuration yaml file you added the corresponding lines as mentioned in the "how to"?

schabau, 2025-07-15:
But as I said, I have the impression that the radar doesn't work on this model (ZG-204ZS). Without active movement, no presence is detected.
If I had bought it from AliExpress, I would have complained. That always worked well. But I bought it from kleinanzeigen.de, so that's bad luck.

Best regards,
Detlef Paschke

rinyheuvelman, 2025-07-15:
I bought them from AliExpress and only one of has functionality of a mmWave sensor, couple of others exactly like yours: only motion detection and no mmWave options or functionality at all. All added via ZHA without quirks or anything. opening one of the "fake/Hobeian" ones (ZG-204ZS with no options for mmWave) did actually had a mmWave sensor present on the board. (See picture above)
Why it doesn't show up after being added, I don't know...
I was under the impression that the missing options are available if the sensor was added via Z2M. Others claim they got it up and running by using a quirk. Sadly enough I don't have a clue about how to make/use/install quirks to ZHA and Home Assistant. I would like to try and see if I could make them work properly by adding them via ZHA after adding a working quirk to ZHA. So I'll try your solution and will get back to you with the results. A little update from AliExpress: I have returned one for free and another one is refunded.

closed 2026-01-20 by github-actions[bot]