zigpy/zha-device-handlers #4217: [BUG] Quirks v2 implementation of Inovelli VZM35-SM fails on some firmware versions

stephenjust, 2025-07-31:
### Bug description

In #3232, the implemenation for VZM35-SM migrated over to Quirks v2.

When attempting to import my diagnostics file into zha for testing, it's failing because the quirk expects a cluster with ID 3 to be present, but it appears that only certain firmware versions will have this cluster present. It appears that at least some part of the Quirks v2 code can't handle missing clusters and so an exception is being raised when initializing the quirk.

### Steps to reproduce

In zha repository with latest dependencies, run `python -m tools.import_diagnostics zha-fe3d142bb254218f191475085bd9eaa1-Inovelli VZM35-SN-29f670420b61de34b51e0ebf3bac38ef.json`

### Expected behavior

Diagnostics file should import successfully (meaning the quirk gets applied successfully)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Diagnostics information

[zha-fe3d142bb254218f191475085bd9eaa1-Inovelli VZM35-SN-29f670420b61de34b51e0ebf3bac38ef.json](https://github.com/user-attachments/files/21522973/zha-fe3d142bb254218f191475085bd9eaa1-Inovelli.VZM35-SN-29f670420b61de34b51e0ebf3bac38ef.json)

### Logs

<details><summary>Logs</summary>

```python
```
2025-07-30 22:24:55 STEPHEN-PC __main__[13257] WARNING Failed to import /mnt/c/Users/steph/Downloads/zha-fe3d142bb254218f191475085bd9eaa1-Inovelli VZM35-SN-29f670420b61de34b51e0ebf3bac38ef.json
Traceback (most recent call last):
  File "/home/stephen/zha/tools/import_diagnostics.py", line 212, in main
    zigpy_device = zigpy_device_from_diagnostics(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/stephen/zha/tools/import_diagnostics.py", line 192, in zigpy_device_from_diagnostics
    return zigpy_device_from_device_data(app, zha_data, patch_cluster)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/stephen/zha/tests/common.py", line 382, in zigpy_device_from_device_data
    device = quirks_get_device(device)
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/stephen/zha/venv/lib/python3.12/site-packages/zigpy/quirks/__init__.py", line 43, in get_device
    return _DEVICE_REGISTRY.get_device(device)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/stephen/zha/venv/lib/python3.12/site-packages/zigpy/quirks/registry.py", line 120, in get_device
    return entry.create_device(device)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/stephen/zha/venv/lib/python3.12/site-packages/zigpy/quirks/v2/__init__.py", line 571, in create_device
    return CustomDeviceV2(device.application, device.ieee, device.nwk, device, self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/stephen/zha/venv/lib/python3.12/site-packages/zigpy/quirks/v2/__init__.py", line 108, in __init__
    replace_meta(self)
  File "/home/stephen/zha/venv/lib/python3.12/site-packages/zigpy/quirks/v2/__init__.py", line 223, in __call__
    self.remove(device)
  File "/home/stephen/zha/venv/lib/python3.12/site-packages/zigpy/quirks/v2/__init__.py", line 207, in __call__
    endpoint = device.endpoints[self.endpoint_id]
               ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
KeyError: 3
```
```

</details>

puddly, 2025-07-31:
I believe https://github.com/zigpy/zha-device-handlers/blob/81e543a892137dbfa5efd5b426a2049d7d13181f/zhaquirks/inovelli/VZM35SN.py#L12-L13 can be removed, since `.replace_cluster_occurrences(InovelliVZM35SNCluster)` takes care of it.

stephenjust, 2025-08-02:
FYI that this prevents the ZHA plugin from loading on HA 2025.8.0b1, and there's no way to remove devices when the plugin fails to load - so this may be a beta blocking bug.

```
Logger: homeassistant.config_entries
Source: config_entries.py:749
First occurred: 6:33:51 PM (3 occurrences)
Last logged: 6:37:34 PM

Error setting up entry SONOFF Zigbee 3.0 USB Dongle Plus V2 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 137, in async_setup_entry
    async with radio_mgr.connect_zigpy_app() as app:
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/contextlib.py", line 214, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/radio_manager.py", line 183, in connect_zigpy_app
    app = await self.radio_type.controller.new(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        app_config, auto_form=False, start_radio=False
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 257, in new
    await app._load_db()
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 129, in _load_db
    await self._dblistener.load()
  File "/usr/local/lib/python3.13/site-packages/zigpy/appdb.py", line 684, in load
    device = zigpy.quirks.get_device(device)
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/__init__.py", line 43, in get_device
    return _DEVICE_REGISTRY.get_device(device)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/registry.py", line 120, in get_device
    return entry.create_device(device)
           ~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/v2/__init__.py", line 571, in create_device
    return CustomDeviceV2(device.application, device.ieee, device.nwk, device, self)
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/v2/__init__.py", line 108, in __init__
    replace_meta(self)
    ~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/v2/__init__.py", line 223, in __call__
    self.remove(device)
    ~~~~~~~~~~~^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/v2/__init__.py", line 207, in __call__
    endpoint = device.endpoints[self.endpoint_id]
               ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
KeyError: 3
```

TheJulianJES, 2025-08-02:
@stephenjust for the above comment ^:
Also, to give a quick tutorial on how to load custom quirks, either search that on the web or add this to your HA `configuration.yaml`:
```yaml
zha:
  custom_quirks_path: /config/zha_custom_quirks
```

Then create a `zha_custom_quirks` folder in your `/config` directory (this assumes you're running HA OS). Then, create a `.py` file and put the quirk from my first comment in there. HA should load again, as that removes the failing cluster replacement for your non-existent endpoint 3.
I'd be very curious to hear about your results with the events (device automation triggers with double, quadruple, quintuple press and so on). This will determine what fix we should implement.

stephenjust, 2025-08-02:
Can confirm that all three code versions allow zha to load. Only the third option allows automation to work:

```python
"""VZM35-SN Fan Switch."""

from zigpy.profiles import zha
from zigpy.quirks.v2 import QuirkBuilder
from zigpy.zcl import ClusterType

from zhaquirks.inovelli import INOVELLI_AUTOMATION_TRIGGERS, InovelliVZM35SNCluster

(
    QuirkBuilder("Inovelli", "VZM35-SN")
    .replaces_endpoint(1, device_type=zha.DeviceType.DIMMABLE_LIGHT)
    .replace_cluster_occurrences(InovelliVZM35SNCluster)
    .replaces_endpoint(3, device_type=zha.DeviceType.DIMMER_SWITCH)
    .replaces(InovelliVZM35SNCluster, endpoint_id=2, cluster_type=ClusterType.Client)
    .replaces(InovelliVZM35SNCluster, endpoint_id=3, cluster_type=ClusterType.Client)
    .device_automation_triggers(INOVELLI_AUTOMATION_TRIGGERS)
    .add_to_registry()
)
```

stephenjust, 2025-08-02:
Diagnostics with the edited quirk, in case you want to generate a test data file:
[zha-fe3d142bb254218f191475085bd9eaa1-Inovelli VZM35-SN-602f66fb2b092186f1a9f6a95e59d1d0 (2).json](https://github.com/user-attachments/files/21556342/zha-fe3d142bb254218f191475085bd9eaa1-Inovelli.VZM35-SN-602f66fb2b092186f1a9f6a95e59d1d0.2.json)

stephenjust, 2025-08-02:
That sounds like it's related to https://inovelli.com/collections/smart-home/products/zigbee-blue-series-smart-fan-light-canopy-module - I don't have one of these, so if cluster 3 is in fact related to binding, I won't be able to verify the behavior doesn't regress.

closed 2025-08-02 by TheJulianJES