zigpy/zha-device-handlers #4218: [BUG] Scene switches Tuya TS0043 TS0044 do not have entities or controls

Mazvy, 2025-07-31:
### Bug description

MOES 3 and 4 gang switches are incorrectly handled.
There are no switch entities, instead for each button there is a battery entity that as far as I can tell is both incorrect in terms of value and nor should it be this way

There are no entities to listen for states like "button is pressed", etc.

TS0044 _TZ3000_wkai4ga5
[image]

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0001",
        "0x0006"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0001",
        "0x0006"
      ],
      "output_clusters": []
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0001",
        "0x0006"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZ3000_wkai4ga5",
  "model": "TS0044",
  "class": "zhaquirks.tuya.ts0044.TuyaSmartRemote0044TO"
}
```

TS0043 _TZ3000_w8jwkczz
[image]
```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0001",
        "0x0006"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0001",
        "0x0006"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZ3000_w8jwkczz",
  "model": "TS0043",
  "class": "zhaquirks.tuya.ts0043.TuyaSmartRemote0043TO"
}
```

### Steps to reproduce

-

### Expected behavior

-

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

MattWestb, 2025-07-31:
Your device is one remote control and its **sending commands (as events, see in the logbook) to the system** that you can as triggering for automatons.
If its being one outlet you is getting one "switch" in the GUI that you can using for **sending commands to the device** for truing on and off.
The only strange things i can see is that some have implanting battery / power cluster on all end points so you is getting multiple battery status in the GUI.

TheSoulshifter, 2026-08-16:
It seems that this is new. I remember writing automations using the UI, but now I can’t.

dyfrgi, 2026-09-08:
Step by step guide by human:

1. Create Automation -> Create new automation
2. Add Trigger
3. Select the "By type" tab (not the default of "By target")
4. Scroll down to Generic, and pick Device, then click Device on the right.
5. Select the switch in the top dropdown. The lower dropdown of Trigger should populate.
6. Select the specific event you want to listen to.
7. Do the rest of the setup (conditions, actions)

It should look like this when you are done:

[image]

It would be really nice to have these show up in some better way, though. For the Hue integration with Lutron switches, I have an "Event received" option under "By target", which is at least a little better, though it would be pretty awkward if you had many such buttons. I'm not sure if there's some better way to handle this sort of device so that I can have the device show up as its own set of entities under "By target".