zigpy/zha-device-handlers #4227: Update Legrand Switch with Neutral signature

beurle, 2025-08-02:
### Problem description

Firmware updates (v44-v65) have changed the signature

### Solution description

Update zhaquirks/legrand/switch.py

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 1,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4129,
    "maximum_buffer_size": 89,
    "maximum_incoming_transfer_size": 63,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 63,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000f",
        "0xfc01"
      ],
      "output_clusters": [
        "0x0000",
        "0x0005",
        "0x0006",
        "0x0019",
        "0xfc01"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0066",
      "input_clusters": [
        "0x0021"
      ],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Light switch with neutral",
  "class": "zigpy.device.Device"
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

puddly, 2025-08-02:
Let's migrate this device to quirks v2, which doesn't require a signature to match and instead will work with model and manufacturer name alone. Can you open a PR to do this?

beurle, 2025-08-03:
@puddly You have already done this on #4214 ?

There is another to do for the dimmer if you have the inclination #4205

beurle, 2025-11-03:
This appears to be fixed in 2025.10. It is detected by 'zigpy.quirks.v2.CustomDeviceV2'