zigpy/zha-device-handlers #4229: [Device Support Request] Shelly 2PM Gen4

daniel-sc, 2025-08-03:
### Problem description

When connecting a Shelly 2PM Gen4 via Zigbee/ZHA, the device is recognized, but incorrectly shows "tilt" actions instead of "lift". The manual zigbee commands "up_upen", "down_close" and "stop" seem to work fine.

### Solution description

The device should be correctly functioning and recognized.

### Screenshots/Video

[image]

### Diagnostics information

[zha-aad025e989f015c4d39d595af107cda0-Shelly 2PM-5a1d859bd75f93849530f821dea04a39.json](https://github.com/user-attachments/files/21565092/zha-aad025e989f015c4d39d595af107cda0-Shelly.2PM-5a1d859bd75f93849530f821dea04a39.json)

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 5264,
    "maximum_buffer_size": 108,
    "maximum_incoming_transfer_size": 1613,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 1613,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0102"
      ],
      "output_clusters": []
    },
    "239": {
      "profile_id": "0xc001",
      "device_type": "0x2001",
      "input_clusters": [
        "0xfc01",
        "0xfc02"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Shelly",
  "model": "2PM",
  "class": "zigpy.device.Device"
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

Hedda, 2025-08-06:
FYI, while it does not sound related to your issue you still might to also be interested in this other Shelly device request -> https://github.com/zigpy/zha-device-handlers/issues/4125

BernardoCBranco, 2025-08-19:
@daniel-sc I don't believe you have the latest firmware, could it be related to this?

[image]

(https://shelly-api-docs.shelly.cloud/gen2/changelog/)

With HA 2025.8.2 and firmware version 1.7.0, it appears fine on mine.
[image]

shangri26199, 2025-08-20:
I can confirm this worked for me and fixed the buttons directly added by ZHA. Firmware 1.7.0 updatet today.
I'm still unable to set any percentage on lift or tilt, but this may be an me issue.

BernardoCBranco, 2025-08-20:
@shangri26199 You should be able to change the lift manually through the card or through an automation as you can see below. It doesn't show for you?

- **Manual**

[image]

- **Automation**

[image]

BernardoCBranco, 2025-08-20:
@shangri26199 Have you calibrated the device through the shelly app?

[image]

daniel-sc, 2025-08-20:
I can confirm that after upgrading the firmware and re-pairing via zigbee and doing a calibration it works! 😃

Hedda, 2025-08-24:
> Tilt and lift position aren't showing in ZHA either. To open the tilt with ZHA, I need to send a Zigbee cluster or group command with tilt percentage value 0. However, in 50% of cases it doesn't work and instead of opening, the device loses tilt calibration. Manually opening the tilt doesn't work, it needs to be fully closed to have tilt/slat back.

@Gor-Lami I wonder if that tilt position and lift position issue could be related to this code pull request that @jeverley is working on for Home Assistant core here?:

- https://github.com/home-assistant/core/pull/142534

jeverley, 2025-08-24:
> @Gor-Lami I wonder if that tilt position and lift position issue could be related to this code pull request that @jeverley is working on for Home Assistant core here?:
>
> - [...]

That PR is removing some non ZigBee attributes that were previously part of the ZHA integration.

Wouldn't be the root cause in this case, this issue looks more similar to what was addressed in this PR https://github.com/zigpy/zha-device-handlers/pull/4052

Are you able to confirm the value being returned in the `window_covering_type` attribute on the cover cluster?

teodor-hristov, 2025-08-24:
> > > Tilt and lift position aren't showing in ZHA either. To open the tilt with ZHA, I need to send a Zigbee cluster or group command with tilt percentage value 0. However, in 50% of cases it doesn't work and instead of opening, the device loses tilt calibration. Manually opening the tilt doesn't work, it needs to be fully closed to have tilt/slat back.
> > 
> > @Gor-Lami I wonder if that tilt position and lift position issue could be related to this code pull request that @jeverley is working on for Home Assistant core here?:
> > 
> > - https://github.com/home-assistant/core/pull/142534
> 
> That PR is removing some non ZigBee attributes that were previously part of the ZHA integration.
> 
> Wouldn't be the root cause in this case, this issue looks more similar to what was addressed in this PR https://github.com/zigpy/zha-device-handlers/pull/4052
> 
> Are you able to confirm the value being returned in the `window_covering_type` attribute on the cover cluster?
> 
> Edit: I can see mention in the Z2M thread of the [1.7.0 firmware changing the device type from Shutter to Rollershade.](https://github.com/Koenkk/zigbee2mqtt/issues/28012#issuecomment-3155712095)
> 
> ZHA is following the ZigBee spec when enabling functionality and Rollershade is lift only.
> 
> Shelly should report the type 'Tilt_blind_tilt_and_lift' if the device supports both, it could be overridden in the meantime with a quirk.

    We reported wrong window covering type as shutter, so we fixed it to be roller shade. Now when slat is enabled and calibration is done if you enable the slat the window covering type will change.

   One "custom" thing we do is we change the ieee addr of the device when we do profile change. For example you pair the 2pm as a 2 channel relay then change the profile, we will leave the network and additional pair should be done. If you change again the profile we will join automatically if possible.

Gor-Lami, 2025-08-24:
With firmware 1.7.0 enabled and calibration done, every pairing shows Tilt_blind_tilt_and_lift when slat control is enabled @jeverley 
@Hedda as discussed in the other thread, this seems unrelated indeed.
However, calibration after pairing doesn't have any effect in contrast to what @daniel-sc wrote. A few minutes later, two of my 2PM Gen4 lose connection, show unknown tilt position in the web UI or 50 in ZHA and ignore all WindowCovering commands until fully opened or closed. I wonder if the Zigbee chips are damaged or something. Once I remove the devices from ZHA or disable Zigbee in the Shelly setting, the faulty behavior stops immediately.
Apart from this, I think something is wrong with the tilt percentages in ZHA. You need to send a command go_to_tilt_percentage (id: 0x0008) with 0 to tilt up. But also current_position_tilt_percentage (id: 0x0009) is 0 after using down_close (id: 0x0001). As tilt is actually down in this state, I'd have expected the current_position_tilt_percentage (id: 0x0009) to show a 100 or have it exactly vice versa. This is likely the reason why the buttons are not working.

teodor-hristov, 2025-08-26:
> > _We reported wrong window covering type as shutter, so we fixed it to be roller shade. Now when slat is enabled and calibration is done if you enable the slat the window covering type will change. One "custom" thing we do is we change the ieee addr of the device when we do profile change. For example you pair the 2pm as a 2 channel relay then change the profile, we will leave the network and additional pair should be done. If you change again the profile we will join automatically if possible._
> 
> @teodor-hristov If a user can change the window covering type with configuation then will that not cause issues to zha as the device would normally need to be re-interviewed (which is not something that the end-user can intitiate from the ZHA UI so need to manually re-join/re-pair the device?)?  Would it not be better to if the firmware always always report the window_covering_type attribut as 'Tilt_blind_tilt_and_lift' if the device can support both? If not then might need a zha quirk (zha device handler) that overrides the window_covering_type from the lift only Rollershade type to `Tilt_blind_tilt_and_lift` to make it so that device shows as a generic relay supporting commands for both lift and tilt?

You are right it will be better. We though it will confuse the users. If you think it will work better with ZHA I'm fine to look into it.

Gor-Lami, 2025-09-07:
I had a discussion with the customer support from Shelly once again and recorded a video. The connectivity issue with unknown slat position is under control now. I had to submit an RPC command to clear ZigBee and start the pairing again. http://192.168.X.X/rpc/shelly.zigbeeclear. However, both clearing ZigBee or updating to firmware 1.7.1-beta1 didn't fix the problem with the wrong current percentages. After closing the blinds a second time after pairing, current_position_lift_percentage (id: 0x0008) results in 100 and current_position_tilt_percentage (id: 0x0009) in a 0 which is not making any sense.

Markus87, 2026-05-09:
@Gor-Lami Thanks for the shelly.zigbeeclear url, this solved my issues.

The state of the switch input contacts is not provided via zigbee to home assistant, does anybody know if this is a limitation from the shelly firmware or something that can be fixed on the home home assistant side?