zigpy/zha-device-handlers #4238: [Device Support Request]

brankolazarevic, 2025-08-06:
### Problem description

Need support for dimmable light switch _TZE200_6fjev1mn. Tested the existing quirks for tuya dimmable switches but none of them worked. Of anyone can provide advice on how I can do this myself, please let me know.. is it just a matter of testing different cluster definitions?

### Solution description

Tuya _TZE200_6fjev1mn quirk

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

puddly, 2025-08-06:
Tuya devices are often not really Zigbee and do not use Zigbee clusters or attributes for core functionality: all data is tunneled with a proprietary protocol over the `0xEF00` cluster.

brankolazarevic, 2025-08-06:
How frustrating, I dont have a tuya gateway so cant go down that route but I have been reading about how they work. If I understand correctly, I should be able to send a zcl command to the cluster and assuming the right data is sent, perform an on/off.. my idea is to call the zha_toolkit action from a python script and basically try to brute force the command/data and see if i can turn it off.. any insight you can provide to whether this is a good approach and what the data/command id would look like?

puddly, 2025-08-06:
Sure, give it a try. I'm not able to recall off the top of my head what the protocol specifics are but I think it's a matter of bruteforcing the 8-bit datapoint IDs and data types.

That being said, your device appears to have some standard ZCL clusters so it appears like it *may* work to some extent with standard functionality that ZHA implements, without the need for quirks. Does it not work at all? Or is there functionality that is missing?

brankolazarevic, 2025-08-06:
It doesnt expose any entities except LQI, RSSI.. it does look like someone got it working via z2m... any idea how to adapt this to a zha quirk?

fingerprint: tuya.fingerprint("TS0601", ["_TZE200_6fjev1mn"]),
    model: "TS0601_TZE200_1gang_dimmer",
    vendor: "Tuya",
    description: "1 gang smart dimmer",
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.lightBrightnessWithMinMax(),
        e.voltage(), e.current(), e.power(), e.child_lock(),
        tuya.exposes.backlightModeOffOn(),
    ],
    meta: {
        tuyaDatapoints: [
            [16, "backlight_mode", tuya.valueConverter.onOff],
            [21, "current", tuya.valueConverter.divideBy1000],
            [22, "power", tuya.valueConverter.divideBy10],
            [23, "voltage", tuya.valueConverter.divideBy10],
            [101, "child_lock", tuya.valueConverter.lockUnlock],
            [141, "state", tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [142, "brightness", tuya.valueConverter.scale0_254to0_1000],
            [143, "min_brightness", tuya.valueConverter.scale0_254to0_1000],
            [144, "max_brightness", tuya.valueConverter.scale0_254to0_1000],

        ],
    },

PR #4286 (2025-08-25): Add various Tuya device variants
## Proposed change

## Additional information

Fixes #4077
Fixes #4238
Fixes #4277

## Device diagnostics

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
- [ ] Device diagnostics data has been attached
closed 2025-08-25 by TheJulianJES