zigpy/zha-device-handlers #4260: TS0601 _TZE204_upagmta9 battery issue is back

adaniels21487, 2025-08-13:
Hi Team,
I have several TS0601 _TZE204_upagmta9 devices.
When the team were developing the patch for this there was an issue where ZHA wasn't using the tuya specific protocol. this was causing the battery to drain completly within 12-24 hours.
See here: https://github.com/zigpy/zha-device-handlers/issues/2694#issuecomment-1792059698

This was fixed and working fine for months.
My system upgraded to Core: 2025.3.4 on the week of 21/07. The battery on my _TZE204_upagmta9 went flat. I changed it and the new one was dead within 24 hours.

It looks like this issue has come back.

Please let me know what I can do to assist in troubleshooting.

Thanks,
Aaron

puddly, 2025-08-14:
2025.3.4 is almost six months old. Please upgrade to the latest Core version.

TheJulianJES, 2025-08-14:
Please also use the "bug report" issue template in the future when reporting possible bugs: https://github.com/zigpy/zha-device-handlers/issues/new/choose

adaniels21487, 2025-08-26:
Apologies. you are right.
I have upgraded to Core: 2025.8.2 and the issue persists.

Information from the bug report template:
Custom Quirks: None

<details><summary>Steps to reproduce.</summary>

1. Install Core: 2025.8.2
2. Connect TS0601 _TZE204_upagmta9
3. Battery will be flat in about 18 hours.

</details>

<details><summary>Expected Behavour.</summary>
Battery should last 6-12 months.
</details>

<details><summary>Debug Information.</summary>
[zha-f248dc5cefbb479d2f1b5aaf2f1ea11c-_TZE204_upagmta9 TS0601-eaff3051dc01dcbb04926363a696ff34 (1).json](https://github.com/user-attachments/files/21998333/zha-f248dc5cefbb479d2f1b5aaf2f1ea11c-_TZE204_upagmta9.TS0601-eaff3051dc01dcbb04926363a696ff34.1.json)
</details>

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_upagmta9",
  "model": "TS0601",
  "class": "zhaquirks.tuya.builder.EnchantedDeviceV2"
}
```

</details>

Max-Funy, 2026-11-16:
I’m having the same problem with zigbee2mqtt ; what should I enter there? The CPU does not enter sleep mode, but instead consumes approximately 5 mA when a link cannot be established or when a new route is being sought.