zigpy/zha-device-handlers #4265: [Device Support Request] new IKEA RODRET description - device not correctly configured

riccardor18, 2025-08-14:
### Problem description

I buy 2 IKEA RODRET , fortunately  the 1st was correctly associated to the right quirk, the second not.
after 2 days I discover that the model description of the 1st device is "RODRET dimmer" the 2nd "RODRET wireless dimmer", copied and modified locally the quirk it was correctly configured. 
the 1st device firmware was an older version, the 2nd was already updated to the last version

### Solution description

update model_info adding the new description
   signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=2080
        # device_version=1
        # input_clusters=[0, 1, 3, 4, 32, 4096, 64636]
        # output_clusters=[3, 4, 6, 8, 258, 4096]>
        MODELS_INFO: [(IKEA, "RODRET Dimmer"),(IKEA, "RODRET wireless dimmer")],
        

### Screenshots/Video

[image]

[image]

[Paste/upload your media here]

</details>

### Diagnostics information

[zha-01K2J7RP6X4K67QYKY0V364S0E-IKEA of Sweden RODRET wireless dimmer-eabea81932bcd53325bf6a31338ecdff.json](https://github.com/user-attachments/files/21779776/zha-01K2J7RP6X4K67QYKY0V364S0E-IKEA.of.Sweden.RODRET.wireless.dimmer-eabea81932bcd53325bf6a31338ecdff.json)

[zha-01K2J7RP6X4K67QYKY0V364S0E-IKEA of Sweden RODRET Dimmer-75ecca7adb218b1c0bbd6972e3613e51.json](https://github.com/user-attachments/files/21779778/zha-01K2J7RP6X4K67QYKY0V364S0E-IKEA.of.Sweden.RODRET.Dimmer-75ecca7adb218b1c0bbd6972e3613e51.json)

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

puddly, 2025-08-14:
Interesting! They have the same firmware version as well. Could you make a PR implementing this change?

closed 2025-08-14 by TheJulianJES

MattWestb, 2025-08-15:
The device "name" is set in the "user data" on the chip in the factory and this looks being made with purpece and then can using  the same firmware for more different devices (TF 2 button have the same hardware and 2 firmware and is making 4 different devices). And IKEA have getting some devices shipped with defect data settings (strange chaster in the name) but the device is working normal.

MattWestb, 2026-02-04:
Controllers like remotes is sending events that is being used for automation's and cant self being controlled. So no switches in the GUI then its not one light you can truing on or off.