zigpy/zha-device-handlers #4268: [Device Support Request] HOBEIAN ZG-204ZG

thib5, 2025-08-15:
### Problem description

The device work. But it work as a PIR sensor since i don't have all the option to calibrate mmwave.

### Solution description

I try tu use de zigbee2mqtt code and make a custom quirk but it never applied and at this point i'm not good enought 

You can still check my code : 
```

from __future__ import annotations

from typing import Any, Final

import zigpy.types as t
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    PowerConfiguration,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    RelativeHumidity,
    TemperatureMeasurement,
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaPowerConfigurationCluster,
    TuyaZBE000Cluster,
)
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaMCUCluster

TUYA_CLUSTER_ID: Final = 0xEF00

class TuyaHobeianMotionManufCluster(TuyaMCUCluster):

    dp_to_attribute: dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaZBE000Cluster.ep_attribute,
            "occupancy",
            dp_type=t.Bool,
        ),
        2: DPToAttributeMapping(
            TuyaZBE000Cluster.ep_attribute,
            "motion_sensitivity",
            dp_type=t.uint32,
        ),
        101: DPToAttributeMapping(
            RelativeHumidity.ep_attribute,
            "measured_value",
            dp_type=t.uint32,
            converter=lambda x: x * 100,  # Convert to ZCL format (0.01% units)
        ),
        102: DPToAttributeMapping(
            TuyaZBE000Cluster.ep_attribute,
            "fading_time",
            dp_type=t.uint32,
        ),
        104: DPToAttributeMapping(
            TuyaZBE000Cluster.ep_attribute,
            "humidity_calibration",
            dp_type=t.int32,
        ),
        105: DPToAttributeMapping(
            TuyaZBE000Cluster.ep_attribute,
            "temperature_calibration",
            dp_type=t.int32,
            converter=lambda x: x * 10,  # Convert from 0.1°C to 0.01°C
        ),
        106: DPToAttributeMapping(
            IlluminanceMeasurement.ep_attribute,
            "measured_value",
            dp_type=t.uint32,
        ),
        107: DPToAttributeMapping(
            TuyaZBE000Cluster.ep_attribute,
            "illuminance_interval",
            dp_type=t.uint32,
        ),
        108: DPToAttributeMapping(
            TuyaZBE000Cluster.ep_attribute,
            "indicator_mode",
            dp_type=t.Bool,
        ),
        109: DPToAttributeMapping(
            TuyaZBE000Cluster.ep_attribute,
            "temperature_unit",
            dp_type=t.enum8,
        ),
        110: DPToAttributeMapping(
            PowerConfiguration.ep_attribute,
            "battery_percentage_remaining",
            dp_type=t.uint32,
            converter=lambda x: x * 2,  # Convert to ZCL format (0-200, where 200=100%)
        ),
        111: DPToAttributeMapping(
            TemperatureMeasurement.ep_attribute,
            "measured_value",
            dp_type=t.int32,
            converter=lambda x: x * 10,  # Convert from 0.1°C to 0.01°C (ZCL format)
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
        104: "_dp_2_attr_update",
        105: "_dp_2_attr_update",
        106: "_dp_2_attr_update",
        107: "_dp_2_attr_update",
        108: "_dp_2_attr_update",
        109: "_dp_2_attr_update",
        110: "_dp_2_attr_update",
        111: "_dp_2_attr_update",
    }

class TuyaHobeianMotionCluster(TuyaZBE000Cluster):

    class AttributeDefs(TuyaZBE000Cluster.AttributeDefs):
  

        occupancy: foundation.ZCLAttributeDef = foundation.ZCLAttributeDef(
            id=0x0000, type=t.Bool, access="rp", mandatory=False
        )
        motion_sensitivity: foundation.ZCLAttributeDef = foundation.ZCLAttributeDef(
            id=0x0001, type=t.uint16, access="rw", mandatory=False
        )
        fading_time: foundation.ZCLAttributeDef = foundation.ZCLAttributeDef(
            id=0x0002, type=t.uint32, access="rw", mandatory=False
        )
        indicator_mode: foundation.ZCLAttributeDef = foundation.ZCLAttributeDef(
            id=0x0003, type=t.Bool, access="rw", mandatory=False
        )
        illuminance_interval: foundation.ZCLAttributeDef = foundation.ZCLAttributeDef(
            id=0x0004, type=t.uint16, access="rw", mandatory=False
        )
        temperature_calibration: foundation.ZCLAttributeDef = foundation.ZCLAttributeDef(
            id=0x0005, type=t.int16, access="rw", mandatory=False
        )
        humidity_calibration: foundation.ZCLAttributeDef = foundation.ZCLAttributeDef(
            id=0x0006, type=t.int16, access="rw", mandatory=False
        )
        temperature_unit: foundation.ZCLAttributeDef = foundation.ZCLAttributeDef(
            id=0x0007, type=t.enum8, access="rw", mandatory=False
        )

    def _update_attribute(self, attrid: int, value: Any) -> None:
        
        super()._update_attribute(attrid, value)
        
        
        if attrid == self.AttributeDefs.occupancy.id:
            ias_cluster = self.endpoint.ias_zone
            if ias_cluster:
                # Convert presence to IAS Zone status
                zone_status = 1 if value else 0  # 1 = alarm, 0 = normal
                ias_cluster._update_attribute(
                    ias_cluster.AttributeDefs.zone_status.id, zone_status
                )

class HobeianZG204ZV(CustomDevice):

    signature = {
        MODELS_INFO: [("HOBEIAN", "ZG-204ZV")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,                      # 0x0000
                    PowerConfiguration.cluster_id,         # 0x0001
                    Identify.cluster_id,                   # 0x0003
                    IlluminanceMeasurement.cluster_id,     # 0x0400
                    TemperatureMeasurement.cluster_id,     # 0x0402
                    RelativeHumidity.cluster_id,           # 0x0405
                    IasZone.cluster_id,                    # 0x0500
                    TUYA_CLUSTER_ID,                       # 0xEF00
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster,
                    Identify.cluster_id,
                    IlluminanceMeasurement.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                    IasZone.cluster_id,
                    TuyaHobeianMotionCluster,
                    TuyaHobeianMotionManufCluster,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id],
            }
        }
    }

    device_automation_triggers = {
        ("occupancy", "on"): {
            "device_automation_type": "occupancy",
            "trigger_type": "motion",
        },
        ("occupancy", "off"): {
            "device_automation_type": "occupancy",
            "trigger_type": "no_motion",
        },
    }
```

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Diagnostics information

[Paramètres HomeAssistant (3).json](https://github.com/user-attachments/files/21800402/Parametres.HomeAssistant.3.json)

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

EnderBow, 2025-10-23:
It works in Zigbee2Mqtt as said in issue I was also able to get the quirk to be detected but because this also is my first try at making a quirk it just doesn't let me change the sensitivity or wait time till it turns off. Would be really nice to see an qirk made for this that works.

soosp, 2025-10-31:
@zimme Thank you for this quirk!
May I suggest some changes?
```
    .tuya_number(
        dp_id=104,
        attribute_name="humidity_offset",
        type=t.int16s,
        min_value=-30,
        max_value=30,
        step=1,
        translation_key="humidity_offset",
        fallback_name="Humidity offset",
    )
    .tuya_number(
        dp_id=105,
        attribute_name="temperature_offset",
        type=t.int16s,
        min_value=-2,
        max_value=2,
        step=0.1,
        multiplier=0.1,
        translation_key="temperature_offset",
        fallback_name="Temperature offset",
    )
```
I tested the limit, step and multiplier values and found that the values above are correct.
The `Temperature offset` and `Humidity offset` are common names in Home Assistant and have translations to more languages.

soosp, 2025-10-31:
BTW: If possible I prefer the `mode="box"` setting for offsets, because it make easier to fine tune these values on mobile devices.

soosp, 2025-11-01:
Today I tested the `Temperature unit` setting, but I do not experienced  any effects on my device. But if I set the temperature unit in Home Assistant, the temperature has been converted automatically. Maybe this setting is needed in Tuya ecosystem only?

Mr-Cake, 2025-11-09:
Thanks for this, spent hours trying to find a config that loads. 
After all that the occupancy mmwave sensor hardly works. Might be useful in a room where you are moving but not not alot.

soosp, 2025-11-09:
@zimme 
> Here's an updated quirk with you improvements and some adjustments.

Great! It works well at me too.
 
> Yeah, Temperature unit doesn't really do anything for me either and neither does Led indicator.

The LED indicator is working at me. It flashes for a very-very short time when the sensor is at rest and motion detected. The switch turns this behavior on and off as expected.

> I tried to add mode="box" but it didn't do anything.

Yes, I also tried it, without success.

julianjwong, 2025-11-13:
> Here's an updated quirk with you improvements and some adjustments.
> 
> Yeah, Temperature unit doesn't really do anything for me either and neither does Led indicator.
> 
> temp and humidity offsets works for me too.
> 
> I tried to add mode="box" but it didn't do anything.
> 
> I've tried to understand how the illumiannce interval is working, I think it's just controlling the internals of the sensor and have don't correlate to the reporting. It also seems like the sensor triggers on bigger changes outside the interval.
> 
> ```
>  """Tuya motion sensors - HOBEIAN ZG-204ZV quirk."""
> 
> from zigpy.quirks.v2 import EntityType
> from zigpy.quirks.v2.homeassistant import UnitOfTime
> from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass
> import zigpy.types as t
> from zigpy.zcl.clusters.measurement import OccupancySensing
> 
> from zhaquirks.tuya import TuyaLocalCluster
> from zhaquirks.tuya.builder import TuyaQuirkBuilder
> from zhaquirks.tuya.tuya_sensor import TuyaTempUnitConvert
> 
> 
> class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
>     """Tuya local OccupancySensing cluster."""
> 
> 
> # HOBEIAN ZG-204ZV, 10GHz mmWave presence sensor with temp/humidity/illuminance
> # Sensor data (battery, temperature, humidity, illuminance) reported via standard
> # Zigbee clusters (0x0001, 0x0402, 0x0405, 0x0400) by the device itself
> # Tuya datapoints used for: occupancy detection (DP 1) and device configuration
> # (sensitivity, fading time, sensor offsets, illuminance interval, LED, temp unit)
> (
>     TuyaQuirkBuilder("HOBEIAN", "ZG-204ZV")
>     .applies_to("_TZE200_uli8wasj", "TS0601")
>     .applies_to("_TZE200_grgol3xp", "TS0601")
>     .applies_to("_TZE200_rhgsbacq", "TS0601")
>     .applies_to("_TZE200_y8jijhba", "TS0601")
>     .tuya_dp(
>         dp_id=1,
>         ep_attribute=TuyaOccupancySensing.ep_attribute,
>         attribute_name=OccupancySensing.AttributeDefs.occupancy.name,
>         converter=lambda x: x == 1,
>     )
>     .adds(TuyaOccupancySensing)
>     .tuya_number(
>         dp_id=2,
>         attribute_name="motion_detection_sensitivity",
>         type=t.uint16_t,
>         min_value=0,
>         max_value=19,
>         step=1,
>         translation_key="motion_detection_sensitivity",
>         fallback_name="Motion detection sensitivity",
>     )
>     .tuya_number(
>         dp_id=102,
>         attribute_name="fading_time",
>         type=t.uint16_t,
>         device_class=SensorDeviceClass.DURATION,
>         unit=UnitOfTime.SECONDS,
>         min_value=0,
>         max_value=28800,
>         step=1,
>         translation_key="fading_time",
>         fallback_name="Motion keep time",
>     )
>     .tuya_number(
>         dp_id=104,
>         attribute_name="humidity_offset",
>         type=t.int16s,
>         min_value=-30,
>         max_value=30,
>         step=1,
>         translation_key="humidity_offset",
>         fallback_name="Humidity offset",
>     )
>     .tuya_number(
>         dp_id=105,
>         attribute_name="temperature_offset",
>         type=t.int16s,
>         min_value=-2,
>         max_value=2,
>         step=0.1,
>         multiplier=0.1,
>         translation_key="temperature_offset",
>         fallback_name="Temperature offset",
>     )
>     .tuya_number(
>         dp_id=107,
>         attribute_name="illuminance_interval",
>         type=t.uint16_t,
>         device_class=SensorDeviceClass.DURATION,
>         unit=UnitOfTime.MINUTES,
>         min_value=1,
>         max_value=720,
>         step=1,
>         translation_key="illuminance_interval",
>         fallback_name="Illuminance interval",
>     )
>     .tuya_switch(
>         dp_id=108,
>         attribute_name="led_indicator",
>         entity_type=EntityType.STANDARD,
>         translation_key="led_indicator",
>         fallback_name="LED indicator",
>     )
>     .tuya_enum(
>         dp_id=109,
>         attribute_name="temperature_unit_convert",
>         enum_class=TuyaTempUnitConvert,
>         translation_key="temperature_unit_convert",
>         fallback_name="Temperature unit",
>     )
>     .skip_configuration()
>     .add_to_registry()
> )
> ```

Hey thanks for sharing this. I'm not able to use it because HA is complaining of indenting issues. Are you able to attach as a file?

mawarnes, 2026-01-03:
> [@julianjwong](https://github.com/julianjwong) Thank you for the hint, I tried it and Copilot got it to work. I've been writing code in one way or another for 15 years, but never tried Copilot to debug.
> 
> Solution: I had to add set_replacement block.
> 
> ```
> """Tuya motion sensors - HOBEIAN ZG-204ZV quirk."""
> 
> import zigpy.types as t
> from zigpy.quirks.v2 import EntityType
> from zigpy.quirks.v2.homeassistant import UnitOfTime
> from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass
> from zigpy.zcl.clusters.general import Basic, PowerConfiguration, Identify
> from zigpy.zcl.clusters.measurement import OccupancySensing, IlluminanceMeasurement
> from zigpy.zcl.clusters.security import IasZone
> 
> from zhaquirks.tuya import TuyaLocalCluster
> from zhaquirks.tuya.builder import TuyaQuirkBuilder
> 
> 
> class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
>     """Tuya local OccupancySensing cluster."""
> 
> 
> # HOBEIAN ZG-204ZV, 10GHz mmWave presence sensor with temp/humidity/illuminance
> # Sensor data (battery, temperature, humidity, illuminance) reported via standard
> # Zigbee clusters (0x0001, 0x0402, 0x0405, 0x0400) by the device itself
> # Tuya datapoints used for: occupancy detection (DP 1) and device configuration
> # (sensitivity, fading time, sensor offsets, illuminance interval, LED, temp unit)
> (
>     TuyaQuirkBuilder("HOBEIAN", "ZG-204ZV")
>     .applies_to("_TZE200_uli8wasj", "TS0601")
>     .applies_to("_TZE200_grgol3xp", "TS0601")
>     .applies_to("_TZE200_rhgsbacq", "TS0601")
>     .applies_to("_TZE200_y8jijhba", "TS0601")
>     .tuya_dp(
>         dp_id=1,
>         ep_attribute=TuyaOccupancySensing.ep_attribute,
>         attribute_name=OccupancySensing.AttributeDefs.occupancy.name,
>         converter=lambda x: x in (1, 2),  # broadened converter
>     )
>     .adds(TuyaOccupancySensing)
>     .tuya_number(
>         dp_id=2,
>         attribute_name="motion_detection_sensitivity",
>         type=t.uint16_t,
>         min_value=0,
>         max_value=19,
>         step=1,
>         translation_key="motion_detection_sensitivity",
>         fallback_name="Motion detection sensitivity",
>     )
>     .tuya_number(
>         dp_id=102,
>         attribute_name="fading_time",
>         type=t.uint16_t,
>         device_class=SensorDeviceClass.DURATION,
>         unit=UnitOfTime.SECONDS,
>         min_value=0,
>         max_value=28800,
>         step=1,
>         translation_key="fading_time",
>         fallback_name="Motion keep time",
>     )
>     .tuya_number(
>         dp_id=107,
>         attribute_name="illuminance_interval",
>         type=t.uint16_t,
>         device_class=SensorDeviceClass.DURATION,
>         unit=UnitOfTime.MINUTES,
>         min_value=1,
>         max_value=720,
>         step=1,
>         translation_key="illuminance_interval",
>         fallback_name="Illuminance interval",
>     )
>     .tuya_switch(
>         dp_id=108,
>         attribute_name="led_indicator",
>         entity_type=EntityType.STANDARD,
>         translation_key="led_indicator",
>         fallback_name="LED indicator",
>     )
>     .skip_configuration()
>     .add_to_registry()
>     .set_signature(
>         {
>             1: {
>                 "profile_id": 0x0104,
>                 "device_type": 0x0402,  # IAS Zone
>                 "in_clusters": [
>                     Basic.cluster_id,
>                     PowerConfiguration.cluster_id,
>                     Identify.cluster_id,
>                     IlluminanceMeasurement.cluster_id,
>                     IasZone.cluster_id,
>                     0xEF00,  # Tuya manufacturer cluster
>                 ],
>                 "out_clusters": [],
>             }
>         }
>     )
>     .set_replacement(
>         {
>             1: {
>                 "profile_id": 0x0104,
>                 "device_type": 0x0402,
>                 "in_clusters": [
>                     Basic.cluster_id,
>                     PowerConfiguration.cluster_id,
>                     Identify.cluster_id,
>                     IlluminanceMeasurement.cluster_id,
>                     IasZone.cluster_id,
>                     TuyaOccupancySensing.cluster_id,
>                 ],
>                 "out_clusters": [],
>             }
>         }
>     )
> )
> ```

many thanks for taking the time to post your solution, I've applied it to home assistant and I'm just testing it now, the config is  showing in ZHA

[image]

soosp, 2026-01-05:
> [@julianjwong](https://github.com/julianjwong) Thank you for the hint, I tried it and Copilot got it to work. I've been writing code in one way or another for 15 years, but never tried Copilot to debug.
> 
> Solution: I had to add set_replacement block.
> 

Hello,

Were the offsets intentionally left out?
I am referring to the following:
```
    .tuya_number(
        dp_id=104,
        attribute_name="humidity_offset",
        type=t.int16s,
        min_value=-30,
        max_value=30,
        step=1,
        translation_key="humidity_offset",
        fallback_name="Humidity offset",
    )
    .tuya_number(
        dp_id=105,
        attribute_name="temperature_offset",
        type=t.int16s,
        min_value=-2,
        max_value=2,
        step=0.1,
        multiplier=0.1,
        translation_key="temperature_offset",
        fallback_name="Temperature offset",
    )
```

tekhiegit, 2026-10-26:
> 28\. > I just tried the quirks using [`custom_zha_quirks`](https://community.home-assistant.io/t/how-to-setup-local-zha-quirks/341226)`/zg204zv.py`
>      > Contents
>      > It works to some extent. The configuration values do seem to 'flicker' sometimes (reset to some other value, then changing back to their old value). The `fading_time` certainly works 👍.
>      
>      
>      This quirk works, but an error appears in the log on every reboot:
>      Unexpected exception importing custom quirk 'hobeian-zg204zv'
>      ```
>      Traceback (most recent call last):
>        File "/usr/local/lib/python3.14/site-packages/zhaquirks/__init__.py", line 506, in setup
>          spec.loader.exec_module(module)
>          ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
>        File "<frozen importlib._bootstrap_external>", line 759, in exec_module
>        File "<frozen importlib._bootstrap>", line 491, in _call_with_frames_removed
>        File "/config/custom_zha_quirks/hobeian-zg204zv.py", line 101, in <module>
>          .set_signature(
>           ^^^^^^^^^^^^^
>      AttributeError: 'QuirksV2RegistryEntry' object has no attribute 'set_signature'
>      ```
>      
>      
>          
>            
>          
>      
>            
>          
>      
>          
>        
>      
>      Fixed quirk
>      """Tuya motion sensors - HOBEIAN ZG-204ZV quirk."""
>      
>      import zigpy.types as t
>      from zigpy.quirks.v2 import EntityType
>      from zigpy.quirks.v2.homeassistant import UnitOfTime
>      from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass
>      from zigpy.zcl.clusters.general import Basic, PowerConfiguration, Identify
>      from zigpy.zcl.clusters.measurement import OccupancySensing, IlluminanceMeasurement
>      from zigpy.zcl.clusters.security import IasZone
>      
>      from zhaquirks.tuya import TuyaLocalCluster
>      from zhaquirks.tuya.builder import TuyaQuirkBuilder
>      
>      
>      class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
>          """Tuya local OccupancySensing cluster."""
>      
>      
>      # HOBEIAN ZG-204ZV, 10GHz mmWave presence sensor with temp/humidity/illuminance
>      # Sensor data (battery, temperature, humidity, illuminance) reported via standard
>      # Zigbee clusters (0x0001, 0x0402, 0x0405, 0x0400) by the device itself
>      # Tuya datapoints used for: occupancy detection (DP 1) and device configuration
>      # (sensitivity, fading time, sensor offsets, illuminance interval, LED, temp unit)
>      (
>          TuyaQuirkBuilder("HOBEIAN", "ZG-204ZV")
>          .applies_to("_TZE200_uli8wasj", "TS0601")
>          .applies_to("_TZE200_grgol3xp", "TS0601")
>          .applies_to("_TZE200_rhgsbacq", "TS0601")
>          .applies_to("_TZE200_y8jijhba", "TS0601")
>          .adds(IlluminanceMeasurement)
>          .adds(PowerConfiguration)
>          .adds(IasZone)
>          .tuya_dp(
>              dp_id=1,
>              ep_attribute=TuyaOccupancySensing.ep_attribute,
>              attribute_name=OccupancySensing.AttributeDefs.occupancy.name,
>              converter=lambda x: x in (1, 2),  # broadened converter
>          )
>          .adds(TuyaOccupancySensing)
>          .tuya_number(
>              dp_id=2,
>              attribute_name="motion_detection_sensitivity",
>              type=t.uint16_t,
>              min_value=0,
>              max_value=19,
>              step=1,
>              translation_key="motion_detection_sensitivity",
>              fallback_name="Motion detection sensitivity",
>          )
>          .tuya_number(
>              dp_id=102,
>              attribute_name="fading_time",
>              type=t.uint16_t,
>              device_class=SensorDeviceClass.DURATION,
>              unit=UnitOfTime.SECONDS,
>              min_value=0,
>              max_value=28800,
>              step=1,
>              translation_key="fading_time",
>              fallback_name="Motion keep time",
>          )
>          .tuya_number(
>              dp_id=107,
>              attribute_name="illuminance_interval",
>              type=t.uint16_t,
>              device_class=SensorDeviceClass.DURATION,
>              unit=UnitOfTime.MINUTES,
>              min_value=1,
>              max_value=720,
>              step=1,
>              translation_key="illuminance_interval",
>              fallback_name="Illuminance interval",
>          )
>          .tuya_switch(
>              dp_id=108,
>              attribute_name="led_indicator",
>              entity_type=EntityType.STANDARD,
>              translation_key="led_indicator",
>              fallback_name="LED indicator",
>          )
>          .tuya_number(
>              dp_id=104,
>              attribute_name="humidity_offset",
>              type=t.int16s,
>              min_value=-30,
>              max_value=30,
>              step=1,
>              translation_key="humidity_offset",
>              fallback_name="Humidity offset",
>          )
>          .tuya_number(
>              dp_id=105,
>              attribute_name="temperature_offset",
>              type=t.int16s,
>              min_value=-2,
>              max_value=2,
>              step=0.1,
>              multiplier=0.1,
>              translation_key="temperature_offset",
>              fallback_name="Temperature offset",
>          )
>          .skip_configuration()
>          .add_to_registry()
>      )

Thanks for this, wish I had found this thread rather than asking Gemini how to fix this issue. Wasted 3 hours, then finally got sick of it and just searched myself and found this. Brilliant fix and now my sensor stops triggering when I just walk past a room the other side of a wall to the sensor 👌👏