zigpy/zha-device-handlers #4294: device support request; Philips Hue new button model: RDM005

jjg3000kis2025, 2025-08-27:
Hello,
I just added to my HA  what seems to be a relatively new Philips Hue button model RDM005. According to Philips it is a new version of their first version button RDM001. It is detected and supported partially by ZHA but not completely so there is no way for me to detect events when the button is pressed even if button press events are visible in the logs of the device.
[Product page on Philips Site](https://www.philips-hue.com/en-us/p/smart-button-smart-button/046677605773#overview)
[image]

Thank you. 

Here are the infos from zha_toolkit 'get device information':
zha_toolkit_version: v1.1.27
zigpy_version: 0.82.2
zigpy_rf_version: 0.45.3
ieee_org: update.bouton_hue_cuisine_lumiere_firmware
ieee: 00:17:88:01:0f:bc:68:0b
command: zha_devices
command_data: null
start_time: "2025-08-27T08:39:35.419828+00:00"
errors: []
params:
  dir: 0
  tries: 1
  expect_reply: true
  args: []
  kwargs: {}
  read_before_write: true
  read_after_write: true
devices:
  - ieee: 00:17:88:01:0f:bc:68:0b
    nwk: 38277
    manufacturer: Signify Netherlands B.V.
    model: RDM005
    name: Signify Netherlands B.V. RDM005
    quirk_applied: false
    quirk_class: zigpy.device.Device
    quirk_id: null
    manufacturer_code: 4107
    power_source: Battery or Unknown
    lqi: 160
    rssi: -71
    last_seen: "2025-08-27T10:12:08"
    available: true
    device_type: EndDevice
    signature:
      node_descriptor:
        logical_type: 2
        complex_descriptor_available: 0
        user_descriptor_available: 0
        reserved: 0
        aps_flags: 0
        frequency_band: 8
        mac_capability_flags: 128
        manufacturer_code: 4107
        maximum_buffer_size: 82
        maximum_incoming_transfer_size: 128
        server_mask: 11264
        maximum_outgoing_transfer_size: 128
        descriptor_capability_field: 0
      endpoints:
        "1":
          profile_id: "0x0104"
          device_type: "0x0830"
          input_clusters:
            - "0x0000"
            - "0x0001"
            - "0x0003"
            - "0x1000"
            - "0xfc00"
          output_clusters:
            - "0x0000"
            - "0x0003"
            - "0x0004"
            - "0x0005"
            - "0x0006"
            - "0x0008"
            - "0x0019"
            - "0x1000"
      manufacturer: Signify Netherlands B.V.
      model: RDM005
    active_coordinator: false
    entities:
      - entity_id: button.signify_netherlands_b_v_rdm005_identify
        name: Signify Netherlands B.V. RDM005
      - entity_id: sensor.signify_netherlands_b_v_rdm005_battery
        name: Signify Netherlands B.V. RDM005
      - entity_id: update.signify_netherlands_b_v_rdm005_firmware
        name: Signify Netherlands B.V. RDM005
      - entity_id: sensor.bouton_hue_cuisine_lumiere_battery
        name: Signify Netherlands B.V. RDM005
      - entity_id: update.bouton_hue_cuisine_lumiere_firmware
        name: Signify Netherlands B.V. RDM005
      - entity_id: button.bouton_hue_cuisine_lumiere_identify
        name: Signify Netherlands B.V. RDM005
    neighbors: []
    routes: []
    endpoint_names:
      - name: NON_COLOR_SCENE_CONTROLLER
    user_given_name: Bouton Hue cuisine lumiere
    device_reg_id: 6bb461a289c52dd869c827dddbae1f3b
    area_id: cuisine
selectDeviceFields: false
success: true

PierroVat, 2025-08-30:
Same here for me.
How can i turn it : Supported ?
Thanks !

puddly, 2025-08-30:
Please fill out the issue template, you seem to have deleted it, and attach diagnostics information for the device please.

jjg3000kis2025, 2025-09-04:
Thanks. I added a New request following the template. :)

Asky82, 2026-02-20:
No News here? :(