zigpy/zha-device-handlers #4298: [BUG] MLI Tint remote: Scene buttons no longer working

duckbert, 2025-08-28:
### Bug description

All scene buttons of my three Tint remotes from MLI stopped working after upgrading to Home Assistant 2025.8.3. No events are visible in HA. All other buttons work fine. Resetting the remotes does not solve the issue. I am not sure if this is related to: https://github.com/zigpy/zigpy/issues/1640. The protocol shows no error for the remote. I do however get an error for deconz conbee 2 that I am using: ‘’ Cluster 0x0006 on <DeconzDevice model='Conbee II' manuf='dresden elektronik' nwk=0x0000 ieee=00:21:2e:ff:ff:05:08:5b is_initialized=True> has incorrect direction (got <Direction.Client_to_Server: 0> for <ClusterType.Server: 0> cluster).’’

### Steps to reproduce

1. Open event listener in Home Assistant and listen to zha_event
2. Push any scene buttons on the MLI Tint remote
3. No event is displayed

### Expected behavior

Pushing a scene buttons on the remote should fire the corresponding event. 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Diagnostics information

[zha-7f9b7d23de14eaa2d7b02b152564b20f-MLI ZBT-Remote-ALL-RGBW-62f5f79f2ed3557a9427cdc0a2a61757.json](https://github.com/user-attachments/files/22029635/zha-7f9b7d23de14eaa2d7b02b152564b20f-MLI.ZBT-Remote-ALL-RGBW-62f5f79f2ed3557a9427cdc0a2a61757.json)

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4635,
    "maximum_buffer_size": 127,
    "maximum_incoming_transfer_size": 100,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 100,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0800",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x1000"
      ],
      "output_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    }
  },
  "manufacturer": "MLI",
  "model": "ZBT-Remote-ALL-RGBW",
  "class": "zhaquirks.mli.tint.TintRemote"
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

duckbert, 2025-09-25:
Hi all, has anyone an idea what is causing the issue? The scene buttons not working, renders all three remotes useless to me.

bbollin, 2025-09-28:
I can confirm this issue with MLIs 'ZBT-Remote-ALL-RGBW' and 'tint-Remote-white' remotes.