zigpy/zha-device-handlers #4310: [Device Support Request] TS0202 by _TZ3040_bb6xaihh

porkyoot, 2025-09-05:
### Problem description

Device pairs fine, works for one second then disconnects. The motion sensor flashes red when it detects motion but nothing is being reported. No quirks is being used.

### Solution description

Ideally constant connection with the device. Otherwise just motion being reported.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

TS0202
by _TZ3040_bb6xaihh
Connected via [Generic Zigbee Coordinator (EZSP)](https://callisto.grea.me/config/devices/device/f412109e532d6ff6d354a5dbf1606652)
Firmware: 0x00000048

[image]

</details>

### Diagnostics information

[zha-01JVTAPG30QH034NMQHVTPPWHX-_TZ3040_bb6xaihh TS0202-ab487a28b2a23f7b59017d3d9709569e.json](https://github.com/user-attachments/files/22167496/zha-01JVTAPG30QH034NMQHVTPPWHX-_TZ3040_bb6xaihh.TS0202-ab487a28b2a23f7b59017d3d9709569e.json)

### Device signature

<details><summary>Device signature</summary>

```json

IEEE: a4:c1:38:**REDACTED**
Nwk: 0xfc90
Device Type: EndDevice
LQI: 136
RSSI: -66
Last seen: 2025-09-05T09:21:46
Power source: Battery or Unknown

```

</details>

### Logs

<details><summary>Logs</summary>

[home-assistant_2025-09-05T07-36-24.235Z.log](https://github.com/user-attachments/files/22167498/home-assistant_2025-09-05T07-36-24.235Z.log)

</details>

nschimme, 2026-02-07:
This might be very similar to what I did in https://github.com/zigpy/zha-device-handlers/pull/4727

Can you try out the quirk?

```
from zhaquirks import MotionWithReset
from zhaquirks.tuya import TuyaLocalCluster, TuyaPowerConfigurationCluster2AAA
from zhaquirks.tuya.builder import TuyaQuirkBuilder

# Tuya PIR motion sensor, SNZB-03
(
    TuyaQuirkBuilder("_TZ3000_bb6xaihh", "SNZB-03")
    .applies_to("_TZ3040_bb6xaihh", "TS0202")
    .replaces(MotionWithReset)
    .replaces(TuyaPowerConfigurationCluster2AAA)
    .tuya_enchantment()
    .skip_configuration()
    .add_to_registry()
)
```

rbungee, 2026-09-26:
For _TZ3040_bb6xaihh / TS0202, ZHA may apply MotionWithReset, which can force the sensor to clear after 30 seconds.
Workaround: create a custom quirk named:
ts0202_plain_ias.py
Place it in:
/config/custom_zha_quirks/
(or /homeassistant/custom_zha_quirks/ in the File Editor add-on)
The quirk should use the normal IasZone cluster instead of MotionWithReset. After restarting Home Assistant, the sensor waits for its real clear message instead of being reset after 30 `seconds.``` 
```python
"""Custom ZHA quirk for _TZ3040_bb6xaihh / TS0202.

Purpose:
- Match only this exact Tuya PIR variant.
- Keep the device on the standard IAS Zone cluster.
- Do not use MotionWithReset.
- Do not add or write manufacturer-specific keep_time attributes.
"""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    OnOff,
    Ota,
    PowerConfiguration,
    Time,
)
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class TS0202PlainIas(CustomDevice):
    """Use the TS0202's native IAS Zone clear/motion notifications."""

    signature = {
        MODELS_INFO: [
            ("_TZ3040_bb6xaihh", "TS0202"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            },
        },
    }

```