zigpy/zha-device-handlers #4317: [Device Support Request] ZG-102ZM Contact sensor vibration support

DevLn, 2025-09-07:
### Problem description

Add vibration support. Currently only contact sensor is exposed to HA.

### Solution description

Both contact and vibration sensors exposed as entities in HA

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[https://www.zigbee2mqtt.io/devices/ZG-102ZM.html](ZG-102ZM)

</details>

### Diagnostics information

[zha-01JHABBQTJDT9RXD3Y6ZW300NS-HOBEIAN ZG-102ZM-61c5e785afed16aae51c73383b04b3ac.json](https://github.com/user-attachments/files/22193850/zha-01JHABBQTJDT9RXD3Y6ZW300NS-HOBEIAN.ZG-102ZM-61c5e785afed16aae51c73383b04b3ac.json)

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4742,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0xef00"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "HOBEIAN",
  "model": "ZG-102ZM",
  "class": "zigpy.device.Device"
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python

025-09-07 07:55:47.213 DEBUG (MainThread) [zigpy.zcl] [0x338F:1:0xef00] Received ZCL frame: b'\tg\x02\x00g\x01\x04\x00\x01\x00'
2025-09-07 07:55:47.214 DEBUG (MainThread) [zigpy.zcl] [0x338F:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x09>(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=103, command_id=2, *direction=<Direction.Server_to_Client: 1>)
2025-09-07 07:55:47.221 DEBUG (MainThread) [zigpy.zcl] [0x338F:1:0xef00] Unknown cluster command 2 b'\x00g\x01\x04\x00\x01\x00'
2025-09-07 07:55:47.221 DEBUG (MainThread) [zigpy.listeners] Matcher checkin() and command ZCLHeader(frame_control=FrameControl<0x09>(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=103, command_id=2, *direction=<Direction.Server_to_Client: 1>) b'\x00g\x01\x04\x00\x01\x00' are incompatible
2025-09-07 07:55:47.222 DEBUG (MainThread) [zigpy.zcl] [0x338F:1:0xef00] Received command 0x02 (TSN 103): b'\x00g\x01\x04\x00\x01\x00'
2025-09-07 07:55:47.223 DEBUG (MainThread) [zigpy.zcl] [0x338F:1:0xef00] No explicit handler for cluster command 0x02: b'\x00g\x01\x04\x00\x01\x00'
2025-09-07 07:55:47.236 DEBUG (MainThread) [zigpy.zcl] [0x338F:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl<0x10>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 0>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=103, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 0>)
2025-09-07 07:55:47.237 DEBUG (MainThread) [zigpy.zcl] [0x338F:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

```

</details>

skragzombi666, 2025-10-26:
Would apreciate this too!

Idskov, 2026-03-28:
I submitted a PR for this device.
https://github.com/zigpy/zha-device-handlers/pull/4872

closed 2026-10-28 by TheJulianJES